Azure/Industrial-IoT #1306: Incorrect node is subscribed on aggregation server if the underlying server connects after the publisher.

Haeavar, 2021-07-29:
**Describe the bug**
We are currently using an aggregation OpcUa server (AS) to aggregate multiple underlying OpcUa server (US1, USn) using reverse connect. Each US1 server have its own namespace URIs which will be added to the AS namespace table. This happens only after the first time the US1 has connected to the AS.

If the publisher creates the subscription **before** the US1 has connected to AS, the namespaceUri is resolved to the namespace idx  65535 which is obviously not the correct one. The Idx 65535 comes from the `namespaces.GetIndex(nodeId.NamespaceUri)` and casting the result to `ushort`.

This wrong behavior will result in a wrong subscription. If the US1 connects after the publisher to the AS, the namespace will be added to the namespaceTable. But the publisher never refreshes its namespaceTable (on the opened session). Only a reconnect will solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a published nodes json for an aggregating OpcUa server with one node using the namespaceUri declaration `{ "Id": "nsu=urn:UnderlyingOpcUa:TRUMPF:UAInterfaces/http://trumpf.com/TRUMPF-Interfaces/;s=33" }`
2. Keep the underlying OPCUA server (`UnderlyingOpcUa` in this sample) disconnected
3. Start the Publisher
4. Observe, that the node `s=33` is subscribed with namespace idx `65535` 

**Expected behavior**
The subscription fails because the namespace is not found in the namespace table. As soon as the namespace can be found, the subscription is created.

**Screenshots**
This is the pn.json part:
[image]

And the publisher output:
[image]

**OpcPublisher:**
 - OpcPublisher version 2.8.0.

**Additional context**
I will create an PR to fix this bug.

marcschier, 2021-08-05:
Thanks for the contributed fix @Haeavar !

koepalex, 2021-08-19:
@Haeavar with the PR you submitted is this issue solved and can be closed?