Azure/Industrial-IoT #1308: OPC Publisher doesn't send data in order as in generated by OPC simulation server

sinfergit, 2021-07-30:
I have been trying to retrieve sensor data generated by OPC simulation server (data listed in excel file and read by OPC simulation) in to one of the custom modules in Azure IOT Edge. When the data logged in the console it shows me that data has not been logged in order. Following is the JSON for OPC publisher hosted in iot edge as a module

[image]

Following is the published nodes json in gateway device

[image]

Following is the screenshot of my excel sheet data
[image]

But the OPC publisher will not route the data in to modules in order that starting from anywhere but in order . For an example it sends starting from the row ,value 11 for Tag11 and then again sends the next row which has the value 17 for tag 11. And sometimes sends a batch of data. no proper order. This is not a issue with OPC server simulation since i have tested Simulation server with a standalone OPC client and it gets the data in order. Excel is read by simulation server. Appreciate any help on this.

koepalex, 2021-08-11:
The same payload (e.g. batch of message 1,2 and 3) was over `3581` times in the log, made it hard to analyze. But already shows an error, each message should be only handled once in the pyhton client.

Let's have a look of the raw message content:

Content of message 1,2,3:
```json
[
    {
        "MessageId": "1",
        "MessageType": "ua-data",
        "PublisherId": "opc.tcp://Xx-XXXXXX:62640/IntegrationObjects/ServerSimulator_fb8261f4290d7d30e3bcb407ee8df1f3c5118a77",
        "DataSetWriterGroup": "opc.tcp://Xx-XXXXXX:62640/IntegrationObjects/ServerSimulator_fb8261f4290d7d30e3bcb407ee8df1f3c5118a77",
        "Messages": [
            {
                "DataSetWriterId": "opc.tcp://Xx-XXXXXX:62640/IntegrationObjects/ServerSimulator_5cbc71babe599c907f3721de643687a363f62f5c",
                "SequenceNumber": 1,
                "MetaDataVersion": {
                    "MajorVersion": 1,
                    "MinorVersion": 0
                },
                "Timestamp": "2021-08-10T15:09:12.820889Z",
                "Payload": {
                    "http://opcfoundation.org/ReferenceApplications#s=Tag11": {
                        "Value": 16,
                        "SourceTimestamp": "2021-08-10T15:09:08.5141646Z",
                        "ServerTimestamp": "2021-08-10T15:09:11.909918Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag12": {
                        "Value": 57,
                        "SourceTimestamp": "2021-08-10T15:09:08.5141646Z",
                        "ServerTimestamp": "2021-08-10T15:09:11.914485Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag13": {
                        "Value": 69,
                        "SourceTimestamp": "2021-08-10T15:09:08.5141646Z",
                        "ServerTimestamp": "2021-08-10T15:09:11.914485Z"
                    }
                }
            }
        ]
    },
    {
        "MessageId": "2",
        "MessageType": "ua-data",
        "PublisherId": "opc.tcp://Xx-XXXXXX:62640/IntegrationObjects/ServerSimulator_fb8261f4290d7d30e3bcb407ee8df1f3c5118a77",
        "DataSetWriterGroup": "opc.tcp://Xx-XXXXXX:62640/IntegrationObjects/ServerSimulator_fb8261f4290d7d30e3bcb407ee8df1f3c5118a77",
        "Messages": [
            {
                "DataSetWriterId": "opc.tcp://Xx-XXXXXX:62640/IntegrationObjects/ServerSimulator_5cbc71babe599c907f3721de643687a363f62f5c",
                "SequenceNumber": 2,
                "MetaDataVersion": {
                    "MajorVersion": 1,
                    "MinorVersion": 0
                },
                "Timestamp": "2021-08-10T15:09:13.7317705Z",
                "Payload": {
                    "http://opcfoundation.org/ReferenceApplications#s=Tag11": {
                        "Value": 17,
                        "SourceTimestamp": "2021-08-10T15:09:13.5164298Z",
                        "ServerTimestamp": "2021-08-10T15:09:13.519777Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag12": {
                        "Value": 65,
                        "SourceTimestamp": "2021-08-10T15:09:13.5241853Z",
                        "ServerTimestamp": "2021-08-10T15:09:13.5241853Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag13": {
                        "Value": 36,
                        "SourceTimestamp": "2021-08-10T15:09:13.5241853Z",
                        "ServerTimestamp": "2021-08-10T15:09:13.5241853Z"
[... 176 lines cut ...]
                        "SourceTimestamp": "2021-08-10T15:09:38.5613689Z",
                        "ServerTimestamp": "2021-08-10T15:09:38.5613689Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag13": {
                        "Value": 14,
                        "SourceTimestamp": "2021-08-10T15:09:38.5613689Z",
                        "ServerTimestamp": "2021-08-10T15:09:38.5613689Z"
                    }
                }
            }
        ]
    }
]
```
 
I would  recommend that you change your python client, the method you are using `receive_on_message_input` is marked as _deprecated_ see [API documentation](https://docs.microsoft.com/en-us/python/api/azure-iot-device/azure.iot.device.iothubmoduleclient?view=azure-python#receive-message-on-input-input-name--block-true--timeout-none-).

As summary:

* Receiving of message don't seems to block and handling same message over and over again (could be related to deprecation of the API used)
*  OPC UA Simulation starts with the simulation **before** the connection is fully established, therefor the first message contains  6th value change (16 for tag11) instead of the first one (11 for tag11), this should be checked in the OPC simulation (of it is an issue)
* OPC Publisher 2.8 works as expected

@sinfergit would you agree? 
Could you send me the link to the stackoverflow post, that I add the answer there also?

sinfergit, 2021-08-11:
@koepalex Please give me some time to test it again and will update you within next few hours

sinfergit, 2021-08-12:
@koepalex I updated the python client code as following.

[image]

My requirement is to get one message payload(tag 11,12,13) at once as a data row of three variables and send it to a dashboard chart to be updated. I don't want to sample the data, just get one by one(one record of tag 11,12,13).
Published Nodes json setting file is as follows
[image]

But in the console log even after removing deprecated code lines still it gets me the data points as batches of two and three records. And it is not fixed as to 1 or 2 records of a batch. I just want to get  a once record at a time to fulfill my requirement.
And now it has missed one record from publish as i assume which is value 13 for starting with Tag 11. I have Attached my log file which contains the python module log for OPC data below.

sinfergit, 2021-08-12:
[python_test_module_log.zip](https://github.com/Azure/Industrial-IoT/files/6972471/python_test_module_log.zip)

sinfergit, 2021-08-12:
@koepalex Stackover flow question 
https://stackoverflow.com/questions/68591571/opc-publisher-doesnt-send-data-in-order-as-in-generated-by-opc-simulation-serve/68628907#68628907

sinfergit, 2021-08-12:
the first message batch i got (2 messages)
[
 
    {
        "MessageId":"8",
        "MessageType":"ua-data",
        "PublisherId":"opc.tcp://XXXXX:XXXXX/IntegrationObjects/ServerSimulator_fb8261f4290d7d30e3bcb407ee8df1f3c5118a77",
        "DataSetWriterGroup":"opc.tcp://XXXXX:XXXXX/IntegrationObjects/ServerSimulator_fb8261f4290d7d30e3bcb407ee8df1f3c5118a77",
        "Messages":[
            {
                "DataSetWriterId":"opc.tcp://XXXXX:XXXXX/IntegrationObjects/ServerSimulator_5cbc71babe599c907f3721de643687a363f62f5c",
                "SequenceNumber":8,
                "MetaDataVersion":{
                    "MajorVersion":1,
                    "MinorVersion":0
                },
                "Timestamp":"2021-08-12T02:04:54.7960987Z",
                "Payload":{
                    "http://opcfoundation.org/ReferenceApplications#s=Tag11":{
                        "Value":11,
                        "SourceTimestamp":"2021-08-12T02:04:53.8545182Z",
                        "ServerTimestamp":"2021-08-12T02:04:53.8546134Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag12":{
                        "Value":56,
                        "SourceTimestamp":"2021-08-12T02:04:53.8546134Z",
                        "ServerTimestamp":"2021-08-12T02:04:53.8546134Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag13":{
                        "Value":47,
                        "SourceTimestamp":"2021-08-12T02:04:53.8546134Z",
                        "ServerTimestamp":"2021-08-12T02:04:53.8546134Z"
                    }
                }
            }
        ]
    },
    {
        "MessageId":"9",
        "MessageType":"ua-data",
        "PublisherId":"opc.tcp://XXXXX:XXXXX/IntegrationObjects/ServerSimulator_fb8261f4290d7d30e3bcb407ee8df1f3c5118a77",
        "DataSetWriterGroup":"opc.tcp://XXXXX:XXXXX/IntegrationObjects/ServerSimulator_fb8261f4290d7d30e3bcb407ee8df1f3c5118a77",
        "Messages":[
            {
                "DataSetWriterId":"opc.tcp://XXXXX:XXXXX/IntegrationObjects/ServerSimulator_5cbc71babe599c907f3721de643687a363f62f5c",
                "SequenceNumber":9,
                "MetaDataVersion":{
                    "MajorVersion":1,
                    "MinorVersion":0
                },
                "Timestamp":"2021-08-12T02:04:59.7851714Z",
                "Payload":{
                    "http://opcfoundation.org/ReferenceApplications#s=Tag11":{
                        "Value":12,
                        "SourceTimestamp":"2021-08-12T02:04:58.8700867Z",
                        "ServerTimestamp":"2021-08-12T02:04:58.8700867Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag12":{
                        "Value":32,
                        "SourceTimestamp":"2021-08-12T02:04:58.8700867Z",
                        "ServerTimestamp":"2021-08-12T02:04:58.8700867Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag13":{
                        "Value":14,
                        "SourceTimestamp":"2021-08-12T02:04:58.8700867Z",
                        "ServerTimestamp":"2021-08-12T02:04:58.8700867Z"
                    }
                }
            }
        ]
[... 234 lines cut ...]
                    "MinorVersion":0
                },
                "Timestamp":"2021-08-12T02:04:49.7619904Z",
                "Payload":{
                    "http://opcfoundation.org/ReferenceApplications#s=Tag11":{
                        "Value":20,
                        "SourceTimestamp":"2021-08-12T02:04:48.8456577Z",
                        "ServerTimestamp":"2021-08-12T02:04:48.8456577Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag12":{
                        "Value":36,
                        "SourceTimestamp":"2021-08-12T02:04:48.8456577Z",
                        "ServerTimestamp":"2021-08-12T02:04:48.8456577Z"
                    },
                    "http://opcfoundation.org/ReferenceApplications#s=Tag13":{
                        "Value":64,
                        "SourceTimestamp":"2021-08-12T02:04:48.8456577Z",
                        "ServerTimestamp":"2021-08-12T02:04:48.8456577Z"
                    }
                }
            }
        ]
    }

]

koepalex, 2021-08-12:
If you want only one IoT Hub message containing the values from all 3 tags, you need to set the batch size to 1 (and maybe the batch interval to 1s).

I see in your log, that message id started again by 1 this only happens if the OPC Publisher was restarted, so it is not continously

sinfergit, 2021-08-12:
@koepalex  Can you please advice me on how to set the batch size and batch interval to 1s for OPC publisher? 
Is there any chance that same message id re-appears again and again since OPC UA server simulator re-reads the old values from the excel sheet? for example excelsheet has only 10 records and once done with reading the 10 records, it will again go from the scratch and publisher will get them as the old messages coming back?

koepalex, 2021-08-14:
> Is there any chance that same message id re-appears again and again since OPC UA server simulator re-reads the old values from the excel sheet? for example excelsheet has only 10 records and once [...]

In general no, the OPC Publisher generate a unique message id (auto increment) for an OPC UA endpoint. As long as the publishing job is active and the message id will increase, independent from the actual values. The only chance to reset the message id is to restart the OPC Publisher. I assume the OPC Simulation will not terminate the connection, once all values are read? It should not stop increasing the message id, but it could be an edge case.

koepalex, 2021-08-14:
> I want only one message at a time from OPC publisher to my python module so that i can send them to a dashboard one by one in real time and update the widgets. Can you please advice me on how to set the batch size and batch interval to 1s for OPC publisher?

I think the additional command line arguments `--bs=1 --BatchTriggerInterval=00:00:01.000` will do what you expect.

sinfergit, 2021-08-16:
> In general no, the OPC Publisher generate a unique message id (auto increment) for an OPC UA endpoint. As long as the publishing job is active and the message id will increase, independent from the [...]

I assume what you said about OPC publisher getting reconnected after some time again and again is right. I noticed a error occurring in OPC publisher edge module. will attach the logs. 
I have updated the `--bs=1` in and now getting one message at a time. But now the order is mixed again. will attach the python client log as well. I tried connecting to the OPC simulation server with UAExpert OPC client and it works fine without getting disconnected and sending messges in order. So there shouldn't be a problem with OPC simulation server right?

[python_edge_module_log.txt](https://github.com/Azure/Industrial-IoT/files/6989119/python_edge_module_log.txt)
[OPC_Publisher_log.txt](https://github.com/Azure/Industrial-IoT/files/6989123/OPC_Publisher_log.txt)
[image]

koepalex, 2021-08-16:
@sinfergit please add a timestamp (incl. milliseconds) to the log of the python module.

Currently it looks that the python log and the opc publisher log are not for the same timestamps. Also the values change every second, but in the screenshot you have shared the values should only change every 5 seconds. 

Could you please restart all and create a fresh recording and then provide the support bundle.

Regarding the error, the secure channel was terminated, and could not be established anymore, do you have the server logs why it terminated the secure channel?

sinfergit, 2021-08-17:
@koepalex will update with the latest logs

koepalex, 2021-08-18:
Tested with:
* iot-edge-opc-plc
  * 3 OPC UA nodes, that changes value every second
* OPC Publisher 2.8
  * PubSub format
  * batch size 1
* custom python module

All messages are in order:

```powershell
2021-08-18 15:18:02,577:INFO:{"MessageId":"1","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":2,"SourceTimestamp":"2021-08-18T15:18:02.0621626Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":2,"SourceTimestamp":"2021-08-18T15:18:02.0652411Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":2,"SourceTimestamp":"2021-08-18T15:18:02.0653438Z"}}}]}
2021-08-18 15:18:03,405:INFO:{"MessageId":"2","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":3,"SourceTimestamp":"2021-08-18T15:18:03.0702467Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":3,"SourceTimestamp":"2021-08-18T15:18:03.0704781Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":3,"SourceTimestamp":"2021-08-18T15:18:03.0711061Z"}}}]}
2021-08-18 15:18:04,419:INFO:{"MessageId":"3","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":4,"SourceTimestamp":"2021-08-18T15:18:04.0619217Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":4,"SourceTimestamp":"2021-08-18T15:18:04.0628068Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":4,"SourceTimestamp":"2021-08-18T15:18:04.0629499Z"}}}]}
2021-08-18 15:18:05,408:INFO:{"MessageId":"4","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":5,"SourceTimestamp":"2021-08-18T15:18:05.0690494Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":5,"SourceTimestamp":"2021-08-18T15:18:05.0694024Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":5,"SourceTimestamp":"2021-08-18T15:18:05.0697465Z"}}}]}
2021-08-18 15:18:06,408:INFO:{"MessageId":"5","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":6,"SourceTimestamp":"2021-08-18T15:18:06.0634494Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":6,"SourceTimestamp":"2021-08-18T15:18:06.0636539Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":6,"SourceTimestamp":"2021-08-18T15:18:06.0638014Z"}}}]}
2021-08-18 15:18:07,407:INFO:{"MessageId":"6","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":7,"SourceTimestamp":"2021-08-18T15:18:07.058998Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":7,"SourceTimestamp":"2021-08-18T15:18:07.0592192Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":7,"SourceTimestamp":"2021-08-18T15:18:07.0592639Z"}}}]}
2021-08-18 15:18:08,412:INFO:{"MessageId":"7","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":8,"SourceTimestamp":"2021-08-18T15:18:08.0707358Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":8,"SourceTimestamp":"2021-08-18T15:18:08.070875Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":8,"SourceTimestamp":"2021-08-18T15:18:08.0709642Z"}}}]}
2021-08-18 15:18:09,407:INFO:{"MessageId":"8","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":9,"SourceTimestamp":"2021-08-18T15:18:09.0602238Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":9,"SourceTimestamp":"2021-08-18T15:18:09.0604346Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":9,"SourceTimestamp":"2021-08-18T15:18:09.060459Z"}}}]}
2021-08-18 15:18:10,436:INFO:{"MessageId":"9","MessageType":"ua-data","PublisherId":"Standalone_pyTestDevice_opc-publisher","DataSetClassId":null,"Messages":[{"DataSetWriterId":"opc.tcp://opcplcX:50000_5c758c962f7efab5bb0cf035016611f8d3aee82e","MetaDataVersion":{"MajorVersion":1,"MinorVersion":0},"Payload":{"http://microsoft.com/Opc/OpcPlc/#s=FastUInt1":{"Value":10,"SourceTimestamp":"2021-08-18T15:18:10.0745656Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt2":{"Value":10,"SourceTimestamp":"2021-08-18T15:18:10.0747757Z"},"http://microsoft.com/Opc/OpcPlc/#s=FastUInt3":{"Value":10,"SourceTimestamp":"2021-08-18T15:18:10.0748238Z"}}}]}
```

sinfergit, 2021-08-18:
@koepalex 
[Support bundle for opc and python module.zip](https://github.com/Azure/Industrial-IoT/files/7008500/Support.bundle.for.opc.and.python.module.zip)

Hi i checked again and it seems i get more than 100 values per one second when logged with the time stamp in the python module.
[image]

sinfergit, 2021-08-18:
> Tested with:
>
>     * iot-edge-opc-plc
>
>       * 3 OPC UA nodes, that changes value every second
>
>     * OPC Publisher 2.8
>
>       * PubSub format
>       * batch size 1
>
>     * custom python [...]

Is there any possibility i can look into your setup for the OPC publisher module with configuration and python module because i have been stuck with this for days now and still struggling to find the issue here. Anyway i tried OPC server simulator free edition from here [Integration objects OPC server simulator](https://integrationobjects.com/company/news-events/integration-objects-latest-release-opc-ua-server-simulator-for-opc-ua-real-time-historical-data-simulation/)

And following is a screen shot i took form a OPC client (UAExpert) connected to the mentioned OPC UA Server simulator Following is a screen shot of the client connected to server and recieving the messages

[image]

here the log file for the OPC UA server simulator generated 
[EventLog.log](https://github.com/Azure/Industrial-IoT/files/7008614/EventLog.log)

koepalex, 2021-08-18:
Please find the code I used and verified:
[IoTEdgePythonClient.zip](https://github.com/Azure/Industrial-IoT/files/7008835/IoTEdgePythonClient.zip)

* use Visual Studio Code with `Azure IoT Tools`
* create local a docker volume named `pytest`
* call "Build and Run IoT Edge in Simulator" based on `deployment.debug.template.json`

sinfergit, 2021-08-18:
> 
> 
> Please find the code I used and verified:
> [IoTEdgePythonClient.zip](https://github.com/Azure/Industrial-IoT/files/7008835/IoTEdgePythonClient.zip)
> 
>     * use Visual Studio Code with `Azure IoT Tools` extension
> 
>     * create local a docker volume named `pytest`
> 
>     * create a pn.json in volume `pytest`
> 
>     * call "Build and Run IoT Edge in Simulator" based on `deployment.debug.template.json`
> 
> 
> content of pn.json:
> 
> ```json
> [
>     {
>       "EndpointUrl": "opc.tcp://opcplcX:50000",
>       "UseSecurity": false,
>       "OpcNodes": [
>         { "Id": "ns=2;s=FastUInt1" },
>         { "Id": "ns=2;s=FastUInt2" },
>         { "Id": "ns=2;s=FastUInt3" }
>       ]
>     }
>   ]
> ```

Can you please share the OPC publisher module settings you have used ?

sinfergit, 2021-08-19:
@koepalex Routing was the issue for me. in the first route i have not mentioned the OPC publisher module specifically. After mentioning to route messages from OPC publisher now it seems to work without any issue. 
Thanks a lot.

koepalex, 2021-08-19:
I'm happy that your problem was solved and that I was able to support.

To help also others it would be great if you could tag your stack overflow question also with the tag `azure-industrial-iot` and mark my updated comment as answer.

closed 2021-08-19 by koepalex

sinfergit, 2021-08-20:
> 
> 
> I'm happy that your problem was solved and that I was able to support.
> 
> To help also others it would be great if you could tag your stack overflow question also with the tag `azure-industrial-iot` and mark my updated comment as answer.

Sure will do