Azure/Industrial-IoT #1310: How to use IoT Edge built-in message broker when running OPC Publisher in orchestrated mode?

emulic, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
According to this [document](https://github.com/Azure/Industrial-IoT/tree/main/docs/manual#overview-of-how-standalone-mode-works), when running a OPC Publisher module in orchestrated mode, we can't use built-in IoT Edge routing to route the message to some other module running within the same IoT Edge device or even of the parent IoT Edge device. In particular, I mean the restriction regarding orchestrated mode described in this [image](https://github.com/Azure/Industrial-IoT/blob/main/docs/manual/media/image3.png).

**Describe the solution you'd like**
In our current customer project we have two scenarios which require both:
a) Usage of the built-in IoT Edge message broker, to forward the message received from OPC Publisher module to some other module of the **same IoT Edge device** where the OPC Publisher module is running on, in order to react on this message and to accomplish some scanner driven validation directly on premises.
b) Usage of the built-in IoT Edge message broker **in the parent IoT Edge device** of the Edge device, where the OPC Publisher Module is running on, to enrich the message sent from OPC Publisher module of the child Edge device with some additional data like. I.e. to append the production order ID currently assigned to the production machine where the OPC UA message is coming from.

**Describe alternatives you've considered**
I hope the restriction described in the linked documentation considers previous versions of IIoT and is resolved in the version 2.8. But if this restriction is still there, maybe you can propose some quick workaround (maybe some hint what needs to changed in the code?) to be able to use IoT Edge built-in message broker features when running the OPC publisher in orchestrated mode.

**Additional context**
Since our customer has a large number of OPC UA assets, we decided to use OPC publisher in orchestrated mode and benefit form its discovery feature. Please note, this is a running project and we would really appreciate quick response from your side.

marcschier, 2021-08-05:
Hi @emulic , if you need to customize the iot edge deployment manifest to change there routing you need to make the edge deployment "unmanaged".  This is explained in the doc and requires you to tag the device twin of the gateway with unmanaged: true.  Then you can deploy the your module, plus publisher, twin and discovery modules with a custom manifest through portal or az cli when the edge gateway is installed.

marcschier, 2021-08-05:
Please close if this helps.

hansgschossmann, 2021-08-05:
As an additional info: OPC Publisher in Orchestrated mode is not supported in a nested Edge configuration. We have validated OPC Publisher standalone in nested Edge deployments. If your network configuration allows to reach the IoT Platforms HTTP endpoints this might work, but we have not validated it, since this not a typical ISA-95 network configuration.

emulic, 2021-08-05:
> Hi @emulic , if you need to customize the iot edge deployment manifest to change there routing you need to make the edge deployment "unmanaged". This is explained in the doc and requires you to tag [...]

Hello @marcshier.  Thank you for your response.
I'm actually aware of "unmanaged" tag and I often used it to combine additional custom modules with IIoT modules.
But if the information presented in this [image](https://github.com/Azure/Industrial-IoT/blob/main/docs/manual/media/image3.png) is true, this won't help. My custom manifest can help to define routing between the edge modules but it doesn't help if the publisher module (in form of generated device identities as showed in the "Orchestrated Mode" part of the picture) sends telemetry events directly to IoT Hub in cloud.  This is what I'm seeing in the picture.
Maybe I'm just misunderstanding the picture, but this would bypass the route via other modules running in the same Edge device. And more, this would bypass the route via parent Edge Devices which I have in my scenario.

emulic, 2021-08-05:
> As an additional info: OPC Publisher in Orchestrated mode is not supported in a nested Edge configuration. We have validated OPC Publisher standalone in nested Edge deployments. If your network [...]

Hi @hansgschossmann,
thanks for clarification regarding nested Edge deployment. Assuming we have a flat Edge deployment without Edge nesting, and Publisher is  running in Orchestrated mode, will I be able to configure routing of the messages from Publisher to my custom modules running on the same Edge device?

marcschier, 2021-08-07:
Whether data is sent to iot hub or other modules is determined by the route configuration in the manifest.  The default one is set so messages go directly to cloud.  Custom manifests can change this regardless of the cloud side created and configured device identities for the subscriptions.  The route definition supports wildcards, etc. as described in the official documentation for IoT Edge.

koepalex, 2021-08-16:
@emulic was the information from marc sufficient or do you need further support?

koepalex, 2021-08-23:
Closed due to inactivity, please reopen if necessary

closed 2021-08-23 by koepalex