Azure/Industrial-IoT #1311: connected factory solution accelerator in the architecture diagram

djwinter29, 2021-08-03:
**Is your feature request related to a problem? Please describe.**
No. But I do confused with what is the connected factory solution accelerator in the https://github.com/Azure/Industrial-IoT/blob/main/docs/media/IIoT-Diagram.png and the project https://github.com/Azure/azure-iot-connected-factory.

**Describe the solution you'd like**
Is there any examples of how to deploy a connected factory solution accelerator specified in the https://github.com/Azure/Industrial-IoT/blob/main/docs/media/IIoT-Diagram.png

**Describe alternatives you've considered**

**Additional context**

marcschier, 2021-08-05:
Please see https://www.azureiotsolutions.com/Accelerators.

closed 2021-08-05 by marcschier

djwinter29, 2021-08-05:
I believe this the link you are provided is only an automated deployment for [this ](https://github.com/Azure/azure-iot-connected-factory) project.  The last update of this project is 17 month ago, and the project has use quite a few libraries which is obsoleted and heavily coupled with time series insights v1, I don't think there is a easy way to connected to [Industrial-IoT](https://github.com/Azure/Industrial-IoT/) project as described in the [architecture](https://github.com/Azure/Industrial-IoT/blob/main/docs/media/IIoT-Diagram.png). 

Could you help to direct me the right way to link [Connected factory accelerator](https://github.com/Azure/azure-iot-connected-factory)/[web based deployment](https://www.azureiotsolutions.com/Accelerators)  to [Industrial-IoT](https://github.com/Azure/Industrial-IoT/)