Azure/Industrial-IoT #1312: Missing example for string based OPC-UA addressing

gerhswitsiem, 2021-08-05:
**Sample for string based OPC-UA addressing is missing**
In [this file](https://github.com/Azure/Industrial-IoT/blob/main/docs/modules/publisher.md#configuration-via-configuration-file) there is an example for connecting an OPC-UA server. The addressing in the sample is numeric based. In practise the customer values are string based addressed. Please add an example for this use case.

Sample of the numeric based address which is provided in the current example:
[image]
_Screenshot of OPC-UA Client called "UA Expert" - showing the adress field_

Sample of an string based address:
[image]
_Screenshot of OPC-UA Client called "UA Expert" - showing the adress field_

**To Reproduce**
Steps to reproduce the behavior:
1. Set up an KEPServerEX
2. Connect the OPC Publisher Container with the OPC-UA server of the KEPServerEX product
3. Try to connect a string based address

@cursu

monikavar, 2021-08-06:
For string based OPC UA addressing you would to use "s" like this:
`{ "Id": "ns=2;s=Siemens S7 TCP..." }`

Will update the docs.

Hope this helps!

gerhswitsiem, 2021-08-09:
Hello,

Thanks for the quick response. I tried the new configuration on my docker container. Unfortunatly this recommendation doesn't work. I got the following error:

`[08:29:37 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Error monitoring node ns=2;s=Siemens S7 TCP.PLC_1.Motor_Simulation.M1.Motor_Drehzahl due to BadNodeIdUnknown in subscription opc.tcp://192.168.15.100:49320_7f760f3d0e771397d2ea8f53b1687a0ff55202cc
`
Do you have any idea what the problem could be?

Haeavar, 2021-08-09:
I would recommend using the "nsu=...;s" syntax to prevent failures due to changed namespace index. 

`"nsu=http://trumpf.com/TRUMPF-Interfaces/;s=179"`

@monikavar I would recommend to change this as well in the documentation.

gerhswitsiem, 2021-08-09:
@Haeavar 
Thank you for your reply but I also got an error on this configuration:

`"Id": "nsu=2;s=Siemens S7 TCP.PLC_1.Motor_Simulation.M1.Motor_Drehzahl"`

`[09:29:11 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Error monitoring                                                                                                                         node ns=65535;s=Siemens S7 TCP.PLC_1.Motor_Simulation.M1.Motor_Drehzahl due to BadNodeIdUnknown in subscription opc.                                                                                                                        tcp://192.168.15.100:49320_7f760f3d0e771397d2ea8f53b1687a0ff55202cc
`

And also with this configuration:
`"Id": "nsu=Siemens S7 TCP.PLC_1.Motor_Simulation.M1.Motor_Drehzahl;s=2",`

`[09:34:40 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Error monitoring                                                                                                                         node ns=65535;s=2 due to BadNodeIdUnknown in subscription opc.tcp://192.168.15.100:49320_7f760f3d0e771397d2ea8f53b16                                                                                                                        87a0ff55202cc
`

monikavar, 2021-08-09:
@gerhswitsiem 
`nsu` is the namespace uri. In a server the namespace URIs may not change, but the position of a certain namespace URI in the namespace array may change.

So in your case, you should use something like this:
`"Id": "nsu=http://opcfoundation.org/Quickstarts/ReferenceApplications;s=Siemens S7 TCP..."`

Hope this helps!

gerhswitsiem, 2021-08-09:
> Hello,
>
> Thanks for the quick response. I tried the new configuration on my docker container. Unfortunatly this recommendation doesn't work. I got the following error:
>
> `[08:29:37 WRN [...]

I tried the configuration from the first suggestion again. Anyway, now the configuration works and I can **successfully load the data**. So to summarize this up:

If you want to read data out of an local OPC-UA server the string based addressing should be like this:
`"Id": "ns=2;s=Siemens S7 TCP.PLC_1.Motor_Simulation.M1.Motor_Drehzahl"`

Just a final hint: I tested also the direct connection between the OPC Publisher and a Siemens S7-1500 CPU. On this device the OPC-UA address is also string based. The addresses also includes some quotes like this:

Example:
`"Id": "ns=2;s=""Motor_Simulation"."M1"."Motor_Drehzahl""`

For a successful configuration we have to escape the quotes because we are already in a string object of the json config file. The final configuration should look something like this:
`"Id": "ns=2;s="\"Motor_Simulation\".\"M1\".\"Motor_Drehzahl\""`

I tested the configuration above and it works on the first try. 

Please add this example also in the documentation. Thank you really much for your support.

monikavar, 2021-08-11:
Glad it helped!

Updated the docs as well.

closed 2021-08-11 by monikavar