Azure/Industrial-IoT #1320: OPC Publisher doesn't support Nodes where the NodeClass is Object

nelsonmorais, 2021-08-13:
**Describe the bug**
Note: Using OPC Publisher (latest = 2.8) in Standalone mode.

If an OPC-UA Server exposes an Address Space where under the Objects folder is an Object and that Object has (as a simple example) one Variable as a child, trying to subscribe to the Object on the OPC Publisher using the publishednodes.json configuration file fails to subscribe to the Node. 

However, subscribing to a Variable that also has an hierarchy of child Variable succeeds and writes the value of the root Variable as a Json object on the OPC Publisher value that is sent to the IoT Hub..

This means that we cannot define our Address Space as an hierarchy of Objects with Objects/Variable an then Subscribe to the root Object, instead we would need to define the Address Space as an hierarchy of Variables only, which doesn't seem to be the correct way of defining an Address Space.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a simple Address Space with a Single Object with a child Variable
2. On the publishednodes.json subscribe to the Object
3. Start the OPC Publisher
4. See error: Error monitoring node ns=2;s=<ObjectBrowseName> due to **BadAttributeIdInvalid** in subscription opc.tcp://192.168....

**Expected behavior**
We would expect to be able to subscribe to a root Object and have all the child Objects/Variables serialized to the value sent to the IoT Hub.

**Screenshots**
Sample Address space:
[image]
[image]

publishednodes.json
`
    "OpcNodes": [
      {
        "Id": "nsu=<the namespace>;i=5001",
        "OpcSamplingInterval": 1000,
        "OpcPublishingInterval": 1000,
        "HeartbeatInterval": 3600,
        "SkipFirst": false
      }
    ]
`

The error:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 + Ubuntu (WSL), IoT Edge Runtime running on WSL

**Additional context**
If this is not supported, then please describe in the documentation which NodeClasses are supported and describe the best way to model the Address Space for this type of scenario.

marcschier, 2021-08-13:
Only node classes with a value attribute are supported.  

Events are not yet supported but a preview implementation can be found in the feature/alarmsandevents branch.  Browsing based node selection and subscription is not yet supported.  See #47 for a similar request.

marcschier, 2021-08-13:
#1219

hansgschossmann, 2021-09-07:
@nelsonmorais pls check my proposed solution for this scenario in #1321

nelsonmorais, 2021-09-23:
Hi @marcschier,

Sorry for the late response.
Clear that only Node Classes with a Value attribute are supported. Any plans to support Node Classes of Type Object?
Regarding the Events and the branch feature/alarmsandevents any ETA for this to be released?

@hansgschossmann Thanks, I replied to you comment on that thread.

hansgschossmann, 2021-12-14:
Closing this one and continue on #1321

nelsonmorais, 2021-12-14:
@hansgschossmann This issue and the one reported on #1321 are not the same. Why is this one being closed?

marcschier, 2024-08-19:
Added new API to expand nodes (#2320)

closed 2024-08-19 by marcschier