Azure/Industrial-IoT #1321: OPC Publisher (publishednodes.json) request support for regular expressions on NodeId

nelsonmorais, 2021-08-13:
**Is your feature request related to a problem? Please describe.**
In cases where we need to subscribe to multiple Nodes with the same settings and where a regular expression to identify all those Nodes could be used it would be great if we could specify the regex instead of having to identify each and every one NodeId.

**Describe the solution you'd like**
I'd like to have the option to specify a regular expression to identify the NodeId instead of having to define each one individually.

**Describe alternatives you've considered**
Enter each NodeId, which is not handy if these are many and can be found via a regular expression

**Additional context**
See also https://github.com/Azure/Industrial-IoT/issues/1320 where I also would like to subscribe to NodeId of Object NodeClass

hansgschossmann, 2021-09-07:
@nelsonmorais  although i like the idea you are proposing it is high risk to implement it, since it has certain performance implications due to its non-deterministic result. Imagine you configure "*" at the root of the server or the address space is dynamically changing and many new nodes are added matching the regex.
I would propose you are browsing the address space and build up a list of nodes which you want to publish and configure OPC Publisher with this list of nodes. This is a much more deterministic approach completely under your control. Make sense?

nelsonmorais, 2021-09-23:
Hi @hansgschossmann,
Sorry for the late response.

I see and understand the problem, but I still think this kind of option could be of interest to have it directly implemented on the OPCPublisher instead of us having to implement this on our own. 

To overcome some the performance implication, an option would be to have for regex an extra set of parameters for:
- the interval that the regex expression would be evaluated 
- the max number of nodeids to be returned, to prevent going beyond a certain number (maybe a warning on the log if there's a higher number available on the regex once compared to this max number of nodes)

Would this simplify the challenges you've mentioned?

hansgschossmann, 2021-12-14:
hi @nelsonmorais we discussed internally and the functionality you are looking for (recursive parameterization of publishing) is not something we will support in OPC Publisher.
We recommend that you implement a two step approach to achieve this:
1) browse all nodes under a specific node and generate a publishing configuration for OPC Publisher out of the browse results
2) configure OPC Publisher with this list

closing this issue for now.

nelsonmorais, 2021-12-14:
@hansgschossmann Understood. Thanks for the feedback.

closed 2023-07-03 by marcschier