Azure/Industrial-IoT #1334: Subscribing to complex datatypes

Mkasakka, 2021-08-24:
Currently, the OPC publisher IoT edge module is a great tool for simple data like a sensor value here and there. But if you try to subscribe to a complex data type such as a custom struct, the messages coming from the module are a mixture of json and uadb (or plain uadb depending on the cmd-line arguments). An example of such a message, containing two nodes, one bool, and one custom struct:

[
  {
    "NodeId": "urn:BeckhoffAutomation:Ua:PLC1#s=.gstSecLoadings.bLoadAck",
    "ApplicationUri": "urn:BeckhoffAutomation:TcOpcUaServer",
    "DisplayName": "LoadingReady",
    "Value": {
      "Value": false,
      "SourceTimestamp": "2021-08-24T06:15:30.7699011Z"
    }
  },
  {
    "NodeId": "urn:BeckhoffAutomation:Ua:PLC1#s=.gstMesLatest.arrUimess",
    "ApplicationUri": "urn:BeckhoffAutomation:TcOpcUaServer",
    "DisplayName": "arMessages",
    "Value": {
      "Value": {
        "TypeId": "urn:BeckhoffAutomation:Ua:PLC1#s=%3cStructuredDataType%3e%3aST_CUSTOM__DefaultBinary",
        "Body": "BgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACAe0VIFJjXASwAAABUaW1lb3V0IHdoZW4gd2FpdGluZyBjb25maWd1cmF0aW9uIGZyb20gQ09OTgAAAAACAAAAAQAAAACRAAA="
      },
      "SourceTimestamp": "2021-08-24T06:15:30.7699011Z"
    }
  }
] 
Is there a way to decode the body of the second node in the cloud or in another iot edge module? 
If not, could this be a feature? I'm thinking something on the lines of adding the reading of the DatatypeDictionary from the server at startup, which would bring the information of the DataTypes to the Publisher, and then while configuring the nodes the developer could set the type of the node in the pn.json file. 
 
To reproduce the example message above:
- Define the message encoding to json via the command-line arguments  for the module
- Create a struct-type variable at a PLC
- Publish it at an OPC ua Server
- Subscribe to it
- Cause a change on it

Hope this is the right way to ask/suggest this kind of thing.
Br,
Markus

Mkasakka, 2021-08-25:
Hi, yes you are correct I am using the Publisher in standalone mode. The current version is 2.7 and the output in the initial post is now with the message mode "PubSub".

koepalex, 2021-08-25:
I've tested the OPC Publisher in standalone version 2.8.0 with the complexe data type of iot-edge-opc-plc (with --ctb) and the complexe data type is working:

```json
{
      "MessageId": "379",
      "MessageType": "ua-data",
      "PublisherId": "opc.tcp://localhost:50000_39d40f4aae5f07696bc0737f6ba45a4d4bd63d79",
      "DataSetWriterGroup": "opc.tcp://localhost:50000_39d40f4aae5f07696bc0737f6ba45a4d4bd63d79",
      "Messages": [
        {
          "DataSetWriterId": "opc.tcp://localhost:50000_ad96df769e92a48dd17e03373f81ee07296dbfc4",
          "MetaDataVersion": {
            "MajorVersion": 1,
            "MinorVersion": 0
          },
          "Payload": {
            "http://microsoft.com/Opc/OpcPlc/Boiler#i=15013": {
              "Value": {
                "Temperature": {
                  "Top": 3360,
                  "Bottom": 3365
                },
                "Pressure": 103365,
                "HeaterState": "On_1"
              },
              "SourceTimestamp": "2021-08-25T09:14:05.416079Z"
            }
          }
        }
      ]
    }
``` 

So it could be that the OPC UA stack (complexe data type) can't decode your custom type. Would it be possible that you provide an public available endpoint of the OPC UA server, that we can connect and check the data type definitions?

Mkasakka, 2021-08-25:
Hi, sadly I can not provide access to the server. Maybe these help? The first is the description of the node that I try to subscribe to (the above output is was for a test node, the same problem though). And the second one is the definition of ST_Mess_Latest
[image]
[image]

koepalex, 2021-08-25:
@mregen is this information sufficient or do we require something else?

Mkasakka, 2021-08-25:
Hi, sure thing!
Here is the structure of which the array is composed.
[image]

mregen, 2021-09-02:
Hi @Mkasakka , based on the screenshot I cannot root cause the issue. By any chance, could you provide the Nodeset2 or other model files? Is this a custom model or companion spec?
As its a nested data structure its hard to tell for me where the code fails here.

Mkasakka, 2021-09-15:
Sure anything you need! As I understand it is a custom model, can you elaborate on what file you need?

mregen, 2021-09-21:
Hi @Mkasakka , there might be files with extension *.nodeset2.xml in your plc project. Some are used for companion specs and are included by others, very likely there are some files which describe e.g. `ST_MESSAGE_INFO`.

Mkasakka, 2021-09-22:
Hi, sorry to say this but I do not find any files with that suffix. Could it have something to do with the fact that I am using the Opc ua server from Beckhoff?

marcschier, 2022-11-24:
Close stale issue