Azure/Industrial-IoT #1379: Deadband and synchronous communication support

pratson20, 2021-09-13:
Hi,
I am looking for two major feature for customer which is not available in IIoT solution-

1. Dead-band
2.  Synchronous communication through OPC Publisher.

Can you please guide if this available or not?

Regards,
Prateek

hansgschossmann, 2021-09-15:
@pratson20 thanks for you request. Can you provide a few more specific details on "Synchronous communciation through OPC Publisher"

pratson20, 2021-09-22:
@hansgschossmann  Need to read bulk of OPC UA tags data at every minute. There is one option to read asynchronously that   OPC Publisher supports .
e.g I need to read 1000 tags data every one minute.

pratson20, 2021-09-22:
Can you please if can be achieve dead-band filter in Azure OPC, Absolute and percentage dead-band?

hansgschossmann, 2021-12-15:
This is on our backlog, but I cannot yet give you a date when this will be in the product.

marcschier, 2022-12-12:
Dead band will be in 2.9.0 community preview which is planned to release soon.

marcschier, 2023-01-16:
Tracking as deadband support. Please open new feature request for polling.

marcschier, 2023-01-17:
In 2.9.0 all of datachange filter can be configured in pn.json file.  Including DeadBand value and percent/absolute.  Behavior depends on server.

closed 2023-01-17 by marcschier