Azure/Industrial-IoT #1380: Module Stops working after 630 nodes are added

mgustafson94, 2021-09-13:
**Describe the bug**
As soon as I add ~>630 tags the module stops working. I have checked to make sure are no bad tags. If I add them all in smaller groups, it works, but as soon as I add them all it does not. I have checked the stats of the VM running Azure Edge, and it it is nowhere close to full. 

Steps to Reproduce
1. The OPC publisher is running, then I add the additional nodes to the publishednodes.json file
2. Restart the OPCPublisher
3. Module starts, but no data being collected
4. Error: "Failed to enable subscription"

**Expected behavior**
It should continue to work like normal

**Screenshots**
Logs:

[16:12:34 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : opc.tcp://xxxxxxxx:49320/
  # Ingestion duration                 :    00:00:01:57 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :            118 (1/s)
  # Ingress ValueChanges (from OPC)    :          1,714 (14.58/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :          1,714
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :            117
  # Encoder avg Notifications/Message  :             15
  # Encoder avg IoT Message body size  :          4,148 (2%)
  # Encoder avg IoT Chunk (4 KB) usage :              1
  # Estimated IoT Chunks (4 KB) per day:        171,970
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :            117 (1/s)
  # Connection retries                 :              0

[16:12:49 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] File /appdata/publishednodes.json has changed, reloading...
[16:12:54 INF Root] Module exits...
[16:12:54 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Stopping worker supervisor
[16:12:54 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Stopping worker...
[16:12:54 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Closing subscription opc.tcp://:49320/_
[16:12:54 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'opc.tcp://xxxxxxxx:49320/' unregistered from session '' in state Running
[16:12:55 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Processing cancellation received ...
[16:12:55 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: StandalonePublisher_0, Job: Standalone_T1_OPCPublisher processing completed ...
[16:12:55 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker stopping...
[16:12:55 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker stopped.
[16:12:55 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Worker supervisor successfully stopped
[16:12:55 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Stopping all sessions
[16:12:55 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Succesfully stopped all sessions
[16:12:55 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Stopping Module Host...
[16:12:55 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 1: Module T1_OPCPublisher disconnected due to Client_Close - now Disabled...
[16:12:55 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to xxxx using MqttOverTcp.
[16:12:55 INF Root] Starting module OpcPublisher version 2.7.206.28401.
[16:12:55 INF Root] Initiating prometheus at port 9702/metrics
[16:12:55 INF Root] Prometheus metric server started.
[16:12:55 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[16:12:55 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context.
[16:12:56 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module T1_OPCPublisher reconnected due to Connection_Ok.
[16:12:56 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for T1 - OPCPublisher
[16:12:56 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state.
[16:12:56 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state.
[16:12:57 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started.
[16:12:57 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[16:12:57 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] File /appdata/publishednodes.json has changed, reloading...
[16:12:57 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Read 1 items from published nodes file in 00:00:00.0194993
[16:12:57 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Converted items to jobs in 00:00:00.0565038
[16:12:57 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'StandalonePublisher_0'...
[16:12:57 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker StandalonePublisher_0: {}
[16:12:57 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: StandalonePublisher_0 processing job: Standalone_T1_OPCPublisher, mode: Active
[16:12:57 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'opc.tcp://:49320/' registered/updated in session '' in state Init
[16:12:57 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'opc.tcp://:49320/' registered/updated in session '' in state Init
[... 24 lines cut ...]
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request)
   at Opc.Ua.SessionClient.SetPublishingMode(RequestHeader requestHeader, Boolean publishingEnabled, UInt32Collection subscriptionIds, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Subscription.SetPublishingMode(Boolean enabled)
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices.SubscriptionWrapper.GetSubscription(Session session, SubscriptionConfigurationModel configuration, Boolean activate) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Protocol\src\Services\SubscriptionServices.cs:line 666
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices.SubscriptionWrapper.ReapplyAsync(Session session, Boolean activate) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Protocol\src\Services\SubscriptionServices.cs:line 254
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices.SubscriptionWrapper.ActivateAsync(Session session) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Protocol\src\Services\SubscriptionServices.cs:line 224
[16:13:07 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Failed to activate subscription
Opc.Ua.ServiceResultException: BadSecureChannelIdInvalid
   at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
   at Opc.Ua.SessionClient.SetPublishingMode(RequestHeader requestHeader, Boolean publishingEnabled, UInt32Collection subscriptionIds, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Subscription.SetPublishingMode(Boolean enabled)
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices.SubscriptionWrapper.GetSubscription(Session session, SubscriptionConfigurationModel configuration, Boolean activate) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Protocol\src\Services\SubscriptionServices.cs:line 666
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices.SubscriptionWrapper.ReapplyAsync(Session session, Boolean activate) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Protocol\src\Services\SubscriptionServices.cs:line 254
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices.SubscriptionWrapper.ActivateAsync(Session session) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Protocol\src\Services\SubscriptionServices.cs:line 224
[root@t1 ~]# vmstat
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 1  0      0 10723900   2160 3718080    0    0     1     6    2    2  1  0 99  0  0

**Server
 - Azure Edge device OS: RHEL 7.9
 - OPCPublisher: 2.7.206.28401.

**Additional context**
OPCPublisher is connected to Kepware

koepalex, 2021-09-15:
Please retest with latest official version 2.8.0 (LTS), what is using the latest OPC Foundation Stack NuGet. 
We currently running OPC Publisher 2.8.0 with 20k tags subscribed.

mgustafson94, 2021-09-15:
If we pull from "mcr.microsoft.com/iotedge/opc-publisher:latest" will that put 2.8 on? How do I tell on the IoT Edge that it is running 2.8 vs 2.7?

Matt Gustafson
Configuration Analyst - Manufacturing Technology | Bell
Office: +1-817-280-8935
bellflight.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bellflight.com%2F&data=02%7C01%7Cmgustafson%40bellflight.com%7C4050e27306ec4dcdcc5708d7a35795d4%7C2d5b202c8c074168a55166f570d429b3%7C0%7C0%7C637157468873247591&sdata=LoQ66WmydnTd6pHIrHajSkQgoI15YUhm%2FXafYdvwWXQ%3D&reserved=0>

koepalex, 2021-09-15:
latest tag should point to 2.8.0 if you pulled in the last 2 months

dacolgit, 2021-09-16:
Have you tried to test with publisher 2.8.0?
If you still have the issue could you send the following data:
- configuration of Kepware server (you can export the device configuration of the server as csv file)
- publishednodes.json file
- container create option for the publisher

mgustafson94, 2021-09-16:
I have not yet. Because it is a production box, I have to wait until I get all the proper approvals. That should be done early next week. I will let you know then.

Matt Gustafson
Configuration Analyst - Manufacturing Technology | Bell
Office: +1-817-280-8935
bellflight.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bellflight.com%2F&data=02%7C01%7Cmgustafson%40bellflight.com%7C4050e27306ec4dcdcc5708d7a35795d4%7C2d5b202c8c074168a55166f570d429b3%7C0%7C0%7C637157468873247591&sdata=LoQ66WmydnTd6pHIrHajSkQgoI15YUhm%2FXafYdvwWXQ%3D&reserved=0>

jehona-m, 2021-10-04:
Closed due to inactivity. If the issue reoccurs after trying version 2.8.0, please reopen it.

closed 2021-10-04 by jehona-m