Azure/Industrial-IoT #1387: Discovery Service not working with Nested Edge

lucarv, 2021-09-28:
I have a nested edge setup where the child edge is able to scan a network and find all OPC UA servers. I can see that the scanning works correctly and telemetry is sent to IoT Hub. However, no assets are registered and therefore I can't activate and subsequently browse the Assets.
If I have a non-nested edge concurrently installed in the same network, the discovery process is able to register assets.

hansgschossmann, 2021-09-29:
@lucarv Thank you for reporting this issue.  We do not yet support Nested Edge deployments in Industrial-IoT platform and will fix this bug when we are going to support it.

cristipogacean, 2021-10-06:
Closing since we cannot replicate the issue. The reason for this issue seem to be the upstream connectivity in the nested edge infrastructure.  Please reopen if necessary.

closed 2021-10-06 by cristipogacean