Azure/Industrial-IoT #1389: TLS certificate broken after upgrading of the AKS cluster

emulic, 2021-09-29:
**Describe the bug**
After upgrading of the Kubernetes version the AKS cluster and nodes are running on to 1.19.13, the TLS certificate for the domain of the IIoT microservices (including the Engineering Tool UI) is broken. The Kubernetes Ingress Controller Fake Certificate seems to be used by Ingress Nginx controller:
[image]

The reason for upgrading of the Kubernetes version was action recommendation communicated by Microsoft in a mail from 23rd September 2021:
[image]

The upgrade was started by using the Upgrade option in Azure Portal -> AKSCluster -> Cluster Configuration. The chosen Upgrade scope was "Upgrade control plane + all node pools": 
[image]

After upgrade we received  a notification saying "Failed to upgrade node pool":
 [image]

However, according to the Azure portal, the cluster, the node pool and all nodes are now running with the more recent version 1.19.13:
[image]
[image]
[image]

We suppose the notification was wrong. As mentioned, our issue is invalid TLS certificate provided by ingress nginx controller.

**To Reproduce**
1. Setup an IIoT environment with microservices running in AKS by using IIoT Deployment Tool and Helmet charts.
2. Upgrade the IIoT AKS Cluster by using Azure Portal option AKS Cluster -> Cluster Configuration.
3. Open the Engineering Tool URL (or one of the URLs of the IIoT microservices) in browser. Edge and Chrome don't allow the page to be opened due invalid TLS certificate. Firefox provides an option to define an exception and to open the URLs. I can confirm, all services are running, only TLS certificate seems to be invalid.

**Expected behavior**
The TLS certificate should keep being be valid after upgrading the Kubernetes.

**Additional context**
If the applied approach of the Kubernetes cluster upgrade is not supported for any reasons, please let us know what is the supported approach to close such vulnerabilities?
And maybe you can give us some recommendation how to fix our IIoT environment?

karok2m, 2021-10-05:
@emulic, the approach that you used for the upgrade of the cluster is totally correct. We are using the same mechanism when upgrading our internal clusters and so far that worked without a problem. So that should not be the culprit.

By IIoT Deployment Tool you are referring to [Microsoft.Azure.IIoT.Deployment](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-deploy-aks.md), right ? Or are you deploying it somehow else ? That one does not create an agent pool with VMSS, so the error message is most probably irrelevant, though I have not seen that error when upgrading my clusters from Azure Portal. Can I ask which version of Kubernetes were you using before the upgrade ?

The error is most probably caused by [`cert-manager`](https://github.com/jetstack/cert-manager) that we deploy using their Helm [chart](https://artifacthub.io/packages/helm/cert-manager/cert-manager/1.1.0). cert-manager acts as a certificate manager that takes care of issuing certificates and adding them to the namespace of ingress. `Microsoft.Azure.IIoT.Deployment` creates a `letsencrypt-prod` resource of type `ClusterIssuer` which is then referenced from `azure-industrial-iot-ingress` resource in `azure-industrial-iot` namespace. That one also defines the name of the secret that should contain details of the certificate, it should be called `tls-secret` and can be found in `azure-industrial-iot` namespace. To troubleshoot the `cert-manager` itself you can use their [troubleshooting guide](https://cert-manager.io/docs/faq/troubleshooting/). I would start from checking if all the deployments of `cert-manager` are up and running. We deploy the chart in `cert-manager` namespace, so the command for that will be this:

```bash
$ kubectl get deployments -n cert-manager
NAME                      READY   UP-TO-DATE   AVAILABLE   AGE
cert-manager              1/1     1            1           51d
cert-manager-cainjector   1/1     1            1           51d
cert-manager-webhook      1/1     1            1           51d
```

Then you should check logs of the pods that are associated with those deployments to see if any component of `cert-manager` is reporting any issues. Here is a sample of how to get logs of `cert-manager-webhook` deployment:

```bash
$ kubectl logs deploy/cert-manager-webhook -n cert-manager
W1005 08:40:58.043996       1 client_config.go:608] Neither --kubeconfig nor --master was specified.  Using the inClusterConfig.  This might not work.
I1005 08:40:58.044434       1 webhook.go:57] cert-manager/webhook "msg"="using dynamic certificate generating using CA stored in Secret resource"  "secret_name"="cert-manager-webhook-ca" "secret_namespace"="cert-manager"
I1005 08:40:58.044844       1 server.go:146] cert-manager/webhook "msg"="listening for insecure healthz connections"  "address"=":6080"
I1005 08:40:58.044894       1 server.go:159] cert-manager/webhook "msg"="listening for secure connections"  "address"=":10250"
I1005 08:40:58.045930       1 server.go:185] cert-manager/webhook "msg"="registered pprof handlers"
I1005 08:40:58.049899       1 reflector.go:207] Starting reflector *v1.Secret (1m0s) from external/io_k8s_client_go/tools/cache/reflector.go:156
I1005 08:40:59.090737       1 dynamic_source.go:199] cert-manager/webhook "msg"="Updated serving TLS certificate"
```

The most interesting logs are going to be in `cert-manager` deployment:

```bash
$ kakostan@DE-KAKOSTAN:/mnt/c/dev/misc/20211005.aks_upgrade$ kubectl logs deploy/cert-manager -n cert-manager
I1004 14:04:43.256597       1 start.go:74] cert-manager "msg"="starting controller"  "git-commit"="7fbdd6487646e812fe74c0c05503805b5d9d4751" "version"="v1.1.0"
W1004 14:04:43.256695       1 client_config.go:608] Neither --kubeconfig nor --master was specified.  Using the inClusterConfig.  This might not work.
I1004 14:04:43.261293       1 controller.go:169] cert-manager/controller/build-context "msg"="configured acme dns01 nameservers" "nameservers"=["10.0.0.10:53"]
I1004 14:04:43.284227       1 controller.go:129] cert-manager/controller "msg"="starting leader election"
I1004 14:04:43.284343       1 metrics.go:162] cert-manager/controller/build-context/metrics "msg"="listening for connections on" "address"={"IP":"::","Port":9402,"Zone":""}
I1004 14:04:43.313031       1 leaderelection.go:243] attempting to acquire leader lease  kube-system/cert-manager-controller...
I1004 14:06:03.816264       1 leaderelection.go:253] successfully acquired lease kube-system/cert-manager-controller
I1004 14:06:03.819446       1 controller.go:103] cert-manager/controller "msg"="starting controller" "controller"="certificaterequests-issuer-selfsigned"
I1004 14:06:03.819614       1 controller.go:103] cert-manager/controller "msg"="starting controller" "controller"="issuers"
I1004 14:06:03.819759       1 controller.go:103] cert-manager/controller "msg"="starting controller" "controller"="certificaterequests-issuer-acme"
I1004 14:06:03.819919       1 controller.go:103] cert-manager/controller "msg"="starting controller" "controller"="CertificateTrigger"
...
```

To try to simulate a situation like yours, I've removed `tls-secret` from `azure-industrial-iot` namespace. And after that I could temporarily see a screen similar to yours with `Your connection isn't private` error when accessing Engineering Tool. But after a few seconds `cert-manager` kicked in and created a new `tls-secret` secret in `azure-industrial-iot` namespace and everything went back to normal. So ideally, if `cert-manager` is up and running then you should be provided with a valid certificate after some time.

If the above troubleshooting steps do not help, then please attach the logs of all three deployments in `cert-manager` namespace and let's see if there are any problems there.

emulic, 2021-10-05:
> @emulic, the approach that you used for the upgrade of the cluster is totally correct. We are using the same mechanism when upgrading our internal clusters and so far that worked without a problem. So that should not be the culprit.
> 
> By IIoT Deployment Tool you are referring to [Microsoft.Azure.IIoT.Deployment](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-deploy-aks.md), right ? Or are you deploying it somehow else ? That one does not create an agent pool with VMSS, so the error message is most probably irrelevant, though I have not seen that error when upgrading my clusters from Azure Portal. Can I ask which version of Kubernetes were you using before the upgrade ?
> 
> The error is most probably caused by [`cert-manager`](https://github.com/jetstack/cert-manager) that we deploy using their Helm [chart](https://artifacthub.io/packages/helm/cert-manager/cert-manager/1.1.0). cert-manager acts as a certificate manager that takes care of issuing certificates and adding them to the namespace of ingress. `Microsoft.Azure.IIoT.Deployment` creates a `letsencrypt-prod` resource of type `ClusterIssuer` which is then referenced from `azure-industrial-iot-ingress` resource in `azure-industrial-iot` namespace. That one also defines the name of the secret that should contain details of the certificate, it should be called `tls-secret` and can be found in `azure-industrial-iot` namespace. To troubleshoot the `cert-manager` itself you can use their [troubleshooting guide](https://cert-manager.io/docs/faq/troubleshooting/). I would start from checking if all the deployments of `cert-manager` are up and running. We deploy the chart in `cert-manager` namespace, so the command for that will be this:
> 
> ```shell
> $ kubectl get deployments -n cert-manager
> NAME                      READY   UP-TO-DATE   AVAILABLE   AGE
> cert-manager              1/1     1            1           51d
> cert-manager-cainjector   1/1     1            1           51d
> cert-manager-webhook      1/1     1            1           51d
> ```
> 
> Then you should check logs of the pods that are associated with those deployments to see if any component of `cert-manager` is reporting any issues. Here is a sample of how to get logs of `cert-manager-webhook` deployment:
> 
> ```shell
> $ kubectl logs deploy/cert-manager-webhook -n cert-manager
> W1005 08:40:58.043996       1 client_config.go:608] Neither --kubeconfig nor --master was specified.  Using the inClusterConfig.  This might not work.
> I1005 08:40:58.044434       1 webhook.go:57] cert-manager/webhook "msg"="using dynamic certificate generating using CA stored in Secret resource"  "secret_name"="cert-manager-webhook-ca" "secret_namespace"="cert-manager"
> I1005 08:40:58.044844       1 server.go:146] cert-manager/webhook "msg"="listening for insecure healthz connections"  "address"=":6080"
> I1005 08:40:58.044894       1 server.go:159] cert-manager/webhook "msg"="listening for secure connections"  "address"=":10250"
> I1005 08:40:58.045930       1 server.go:185] cert-manager/webhook "msg"="registered pprof handlers"
> I1005 08:40:58.049899       1 reflector.go:207] Starting reflector *v1.Secret (1m0s) from external/io_k8s_client_go/tools/cache/reflector.go:156
> I1005 08:40:59.090737       1 dynamic_source.go:199] cert-manager/webhook "msg"="Updated serving TLS certificate"
> ```
> 
> The most interesting logs are going to be in `cert-manager` deployment:
> 
> ```shell
> $ kakostan@DE-KAKOSTAN:/mnt/c/dev/misc/20211005.aks_upgrade$ kubectl logs deploy/cert-manager -n cert-manager
> I1004 14:04:43.256597       1 start.go:74] cert-manager "msg"="starting controller"  "git-commit"="7fbdd6487646e812fe74c0c05503805b5d9d4751" "version"="v1.1.0"
> W1004 14:04:43.256695       1 client_config.go:608] Neither --kubeconfig nor --master was specified.  Using the inClusterConfig.  This might not work.
> I1004 14:04:43.261293       1 controller.go:169] cert-manager/controller/build-context "msg"="configured acme dns01 nameservers" "nameservers"=["10.0.0.10:53"]
> I1004 14:04:43.284227       1 controller.go:129] cert-manager/controller "msg"="starting leader election"
> I1004 14:04:43.284343       1 metrics.go:162] cert-manager/controller/build-context/metrics "msg"="listening for connections on" "address"={"IP":"::","Port":9402,"Zone":""}
> I1004 14:04:43.313031       1 leaderelection.go:243] attempting to acquire leader lease  kube-system/cert-manager-controller...
> I1004 14:06:03.816264       1 leaderelection.go:253] successfully acquired lease kube-system/cert-manager-controller
> I1004 14:06:03.819446       1 controller.go:103] cert-manager/controller "msg"="starting controller" "controller"="certificaterequests-issuer-selfsigned"
> I1004 14:06:03.819614       1 controller.go:103] cert-manager/controller "msg"="starting controller" "controller"="issuers"
> I1004 14:06:03.819759       1 controller.go:103] cert-manager/controller "msg"="starting controller" "controller"="certificaterequests-issuer-acme"
> I1004 14:06:03.819919       1 controller.go:103] cert-manager/controller "msg"="starting controller" "controller"="CertificateTrigger"
> ...
> ```
> 
> To try to simulate a situation like yours, I've removed `tls-secret` from `azure-industrial-iot` namespace. And after that I could temporarily see a screen similar to yours with `Your connection isn't private` error when accessing Engineering Tool. But after a few seconds `cert-manager` kicked in and created a new `tls-s [... cut ...]
tls-secret                                      kubernetes.io/tls                     2      69d
```
I tried to check the certificate by following the cert-manager troubleshooting guide you referenced to and executed the following command in all namespaces:
```shell
kubectl get certificate
```
No certificates could be found in any of the namespaces.

Maybe you have an idea, what this could mean? 

As I mentioned, our environment is working again, but we would just like to understand what went wrong, just to be prepared for the case the issue occurs later in the production environment.

emulic, 2021-10-05:
Addendum:
To get the logs you asked for, I stopped the new cluster and started the old one again. After putting all information to my previous post, I wanted to revert back to the new, working one, so I stopped the old cluster and started again the new one.
Now we have the same "Your connection isn't private" issue in the new cluster too. It looks like the issue occurs after restarting the cluster (I assume restart of the old cluster was implicitly done by upgrade).
Maybe I should delete the tls-secret and see if this will automatically trigger creation of the new certificate as it happened in your case?

karok2m, 2021-10-06:
@emulic, can you please use a namespaces version of `kubectl get certificate` command ? Here is the output of that and the `describe` commands for me.

```bash
$ kubectl get certificate -n azure-industrial-iot
NAME         READY   SECRET       AGE
tls-secret   True    tls-secret   4m5s

$ kubectl describe certificate tls-secret -n azure-industrial-iot
Name:         tls-secret
Namespace:    azure-industrial-iot
Labels:       app.kubernetes.io/component=ingress
              app.kubernetes.io/instance=azure-industrial-iot
              app.kubernetes.io/managed-by=Helm
              app.kubernetes.io/name=azure-industrial-iot
              app.kubernetes.io/part-of=azure-industrial-iot
              app.kubernetes.io/version=2.7.206
              helm.sh/chart=azure-industrial-iot-0.3.2
Annotations:  <none>
API Version:  cert-manager.io/v1
Kind:         Certificate
Metadata:
  Creation Timestamp:  2021-10-06T14:58:36Z
  Generation:          1
  Managed Fields:
    API Version:  cert-manager.io/v1
    Fields Type:  FieldsV1
    fieldsV1:
      f:metadata:
        f:labels:
          .:
          f:app.kubernetes.io/component:
          f:app.kubernetes.io/instance:
          f:app.kubernetes.io/managed-by:
          f:app.kubernetes.io/name:
          f:app.kubernetes.io/part-of:
          f:app.kubernetes.io/version:
          f:helm.sh/chart:
        f:ownerReferences:
          .:
          k:{"uid":"05f0c806-2728-4ed5-a19f-e8505e8344fc"}:
            .:
            f:apiVersion:
            f:blockOwnerDeletion:
            f:controller:
            f:kind:
            f:name:
            f:uid:
      f:spec:
        .:
        f:dnsNames:
        f:issuerRef:
          .:
          f:group:
          f:kind:
          f:name:
        f:secretName:
      f:status:
        .:
        f:conditions:
        f:notAfter:
        f:notBefore:
        f:renewalTime:
    Manager:    controller
    Operation:  Update
    Time:       2021-10-06T14:58:36Z
  Owner References:
    API Version:           extensions/v1beta1
    Block Owner Deletion:  true
    Controller:            true
    Kind:                  Ingress
[... 136 lines cut ...]
116s        Normal    Killing            pod/cert-manager-756bb56c5-ggx42                Stopping container cert-manager
116s        Normal    Scheduled          pod/cert-manager-756bb56c5-mrg9x                Successfully assigned cert-manager/cert-manager-756bb56c5-mrg9x to aks-agentpool-24301254-0
115s        Normal    Pulling            pod/cert-manager-756bb56c5-mrg9x                Pulling image "quay.io/jetstack/cert-manager-controller:v1.1.0"
111s        Normal    Pulled             pod/cert-manager-756bb56c5-mrg9x                Successfully pulled image "quay.io/jetstack/cert-manager-controller:v1.1.0" in 4.409094688s
110s        Normal    Created            pod/cert-manager-756bb56c5-mrg9x                Created container cert-manager
110s        Normal    Started            pod/cert-manager-756bb56c5-mrg9x                Started container cert-manager
116s        Normal    SuccessfulCreate   replicaset/cert-manager-756bb56c5               Created pod: cert-manager-756bb56c5-mrg9x
116s        Normal    Killing            pod/cert-manager-cainjector-86bc6dc648-c5wn6    Stopping container cert-manager
116s        Normal    Scheduled          pod/cert-manager-cainjector-86bc6dc648-z5lbs    Successfully assigned cert-manager/cert-manager-cainjector-86bc6dc648-z5lbs to aks-agentpool-24301254-1
115s        Normal    Pulled             pod/cert-manager-cainjector-86bc6dc648-z5lbs    Container image "quay.io/jetstack/cert-manager-cainjector:v1.1.0" already present on machine
114s        Normal    Created            pod/cert-manager-cainjector-86bc6dc648-z5lbs    Created container cert-manager
114s        Normal    Started            pod/cert-manager-cainjector-86bc6dc648-z5lbs    Started container cert-manager
116s        Normal    SuccessfulCreate   replicaset/cert-manager-cainjector-86bc6dc648   Created pod: cert-manager-cainjector-86bc6dc648-z5lbs
116s        Normal    Scheduled          pod/cert-manager-webhook-66b555bb5-lfxfc        Successfully assigned cert-manager/cert-manager-webhook-66b555bb5-lfxfc to aks-agentpool-24301254-0
114s        Normal    Pulled             pod/cert-manager-webhook-66b555bb5-lfxfc        Container image "quay.io/jetstack/cert-manager-webhook:v1.1.0" already present on machine
113s        Normal    Created            pod/cert-manager-webhook-66b555bb5-lfxfc        Created container cert-manager
113s        Normal    Started            pod/cert-manager-webhook-66b555bb5-lfxfc        Started container cert-manager
116s        Normal    Killing            pod/cert-manager-webhook-66b555bb5-r79gs        Stopping container cert-manager
114s        Warning   Unhealthy          pod/cert-manager-webhook-66b555bb5-r79gs        Readiness probe failed: Get "http://10.240.0.23:6080/healthz": dial tcp 10.240.0.23:6080: connect: connection refused
116s        Normal    SuccessfulCreate   replicaset/cert-manager-webhook-66b555bb5       Created pod: cert-manager-webhook-66b555bb5-lfxfc
```

@emulic, since the few logs that I can see are really not enough to understand what is going on in the system, can you please try to restart the whole `cert-manager` and see if it manages to do anything and then collect full logs of all three services and attach them here ? To restart all components of `cert-manager` you can maybe delete all pods from `cert-manager` namespace and let the deployments create new ones. This can be achieved with `kubectl delete --all pods -n cert-manager` command.

Regarding the log in `cert-manager-webhook`, I also have that in my logs, and that does not seem to cause any issues in my case.

emulic, 2021-10-18:
Hi @karok2m.
not yet. We just were able to get everything working again by cleaning the whole cluster and setup it again by executing most of the steps described in this [instructions](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-add-aks-to-ps1.md). I'm trying to find some time slot to try out your recommendations, to avoid disturbing somebody else using the environment.
Please give me some more time, I'll let you know about my findings.

In the meantime, maybe you can try to stop the whole AKS cluster on your side and start it again, just as I did with our new cluster (I simply used stop/start in Azure Portal), to see if the TLS certificate will be broken after restarting like it was in our case?

Maybe you can confirm that our approach to create IIoT microservices AKS cluster by using this [instructions](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-add-aks-to-ps1.md) on top of the resources created by using [Microsoft.Azure.IIoT.Deployment](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-deploy-aks.md) is supported way? Right now I'm not sure if this is the case since the mentioned [instructions](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-add-aks-to-ps1.md) is mentioning only ```deploy.cmd``` and ```deploy.sh``` scripts.

Regarding putting namespace to the ```kubectl get certificate``` command in our old cluster, that's exactly what I did. I have executed the command for all 3 namespaces and none of them had any certificates.

emulic, 2021-11-17:
@karok2m any ideas?

karok2m, 2022-04-12:
@emulic, I've also tried the cluster stop/start scenario that you've mentioned with an AKS cluster which uses virtual machine scale set as agent pool type instead of availability set (that is used by `Microsoft.Azure.IIoT.Deployment`). And in my case everything worked as expected. After all workloads were up and running, I've been able to access Engineering Tool and Swagger pages of the microservices without any issues. Please note that I had to create an AKS cluster that uses virtual machine scale set to be able to stop and start the cluster. For clusters that are backed by availability sets the feature is [not supported](https://docs.microsoft.com/en-us/azure/aks/start-stop-cluster?tabs=azure-cli#limitations). I will be changing `Microsoft.Azure.IIoT.Deployment` to also use  virtual machine scale set.

karok2m, 2022-04-20:
#1663 changed `Microsoft.Azure.IIoT.Deployment` to use  [virtual machine scale set](https://docs.microsoft.com/en-us/azure/virtual-machines/flexible-virtual-machine-scale-sets) instead of [availability set](https://docs.microsoft.com/en-us/azure/virtual-machines/availability-set-overview) as type of agent pool used in AKS cluster. That allows for addition of new node pools after cluster creation and enables start/stop functionality of AKS cluster.

karok2m, 2022-04-29:
@emulic, I am closing this issue as I was not able to reproduce the problem and the current version seems to behave as expected. Please reopen it if you are still having problems with the deployment.

closed 2022-04-29 by karok2m