Azure/Industrial-IoT #1398: Add StyleCop to improve PR request experience and styling consistency

suneetnangia, 2021-09-30:
**Is your feature request related to a problem? Please describe.**
Currently, we do not have a way to allow styling rules to stay with the code, this results in OSS devs ending up with PRs having trailing whitespaces or files without blank lines at the end. Getting those right first time on the local machine will help reducing the unwanted round trips and improve overall dev experience/enforcements.

**Describe the solution you'd like**
If you agree it's a good idea, I can raise a PR to add StyleCop config in the common.props and at least trailing whitespace rule (SA1028) in .editorconfig.

CC @karok2m 

Thanks.

karok2m, 2021-10-01:
@suneetnangia, for the further development we wanted to use a ruleset defined in `.editorconfig` and the StyleCop rules from `stylecop.json` file to have the best possible support to follow the conventions by the IDE. So I was thinking that maybe we should have a a solution-wide `stylecop.json` file. Though, from what I understand, having a solution-wide `stylecop.json` file would actually mean that it should be linked from each project individually. How much different using such `stylecop.json` file would be from your approach of adding the StyleCop configuration in `common.props` file ?

marcschier, 2023-01-16:
PR welcome.  However, closing this issue as it is fundamentals related, which we track internally.

closed 2023-01-16 by marcschier