Azure/Industrial-IoT #1437: Orchestrated Publisher: Container create option "--mm=PubSub" not working.

mwiot, 2021-10-21:
Hi,
I am having trouble getting the container create options to work properly for the publisher module when using the cloud orchestrated solution. Specifically I am trying to get the option "--mm=PubSub" to work, but I am not sure that any of the container create options are being used. 

- How can I get PubSub format when using the orchestrated publisher? Is it not possible to set this in container create options when it is cloud configured? If so, how can the message format be set in the publisher cloud service?
 

Some additional info. Using: 
 - OS: Ubuntu 20.04
 - Edge runtime:  iotedge 1.1.7
 - All-in-one iiot services 2.8.0
 - All edge modules: 2.8.0

Case 1: Use standalone publisher solution, nodes defined in pn.json

Publisher config:
```
{
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "--pf=/appdata/pn.json",
    "--di=60",
    "--to",
    "--aa",
    "--si=10",
    "--ms=262144",
    "--mm=PubSub"
  ],
  "HostConfig": {
    "PortBindings": {
      "62222/tcp": [
        {
          "HostPort": "62222"
        }
      ]
    },
    "Binds": [
      "/var/iiotedge:/appdata"
    ],
    "ExtraHosts": [
      "localhost:127.0.0.1"
    ]
  }
}
```

IOT hub telemetry contains the correct message format  specified in the container create options (pubsub):

```
[IoTHubMonitor] [1:42:57 PM] Message received from [vm-opc-simulator4/publisher]:
[
  {
    "MessageId": "101",
    "MessageType": "ua-data",
    "PublisherId": "opc.tcp://10.0.0.9:50000/_0f38f3b7bb07c7fd6505a9330091ce3660c8a868",
    "DataSetWriterGroup": "opc.tcp://10.0.0.9:50000/_0f38f3b7bb07c7fd6505a9330091ce3660c8a868",
    "Messages": [
      {
        "DataSetWriterId": "opc.tcp://10.0.0.9:50000/_f20e4a77ec212a940d80bb5773ff0ba92a22882a",
        "MetaDataVersion": {
          "MajorVersion": 1,
          "MinorVersion": 0
        },
        "Payload": {
          "nsu=http://microsoft.com/Opc/OpcPlc/;s=SlowUInt1": {
            "Value": 11142,
            "SourceTimestamp": "2021-10-21T13:42:54.4742935Z"
          },
          "nsu=http://microsoft.com/Opc/OpcPlc/;s=SlowUInt2": {
            "Value": 11142,
            "SourceTimestamp": "2021-10-21T13:42:54.4743084Z"
          }
        }
      }
    ]
  }
]
```

Case 2: Use orchestrated publisher , nodes defined in publisher cloud service. 

Publisher config: 
```
{
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "--di=60",
    "--to",
    "--aa",
    "--si=10",
    "--ms=262144",
    "--mm=PubSub"
  ],
  "HostConfig": {
    "PortBindings": {
      "62222/tcp": [
        {
          "HostPort": "62222"
        }
      ]
    },
    "Binds": [
      "/var/iiotedge:/appdata"
    ],
    "ExtraHosts": [
      "localhost:127.0.0.1"
    ]
  }
}
```
IOT Hub telemetry does NOT contain the configured format, PubSub. This is the expected format when using --mm=Samples. 

```
 {
    "NodeId": "http://microsoft.com/Opc/OpcPlc/#s=SlowUInt4",
    "EndpointUrl": "opc.tcp://10.0.0.8:50000/",
    "ApplicationUri": "urn:OpcPlc:aa01c4a97852",
    "DisplayName": "SlowUInt4",
    "Timestamp": "2021-10-21T13:46:30.4780657Z",
    "Status": "Good",
    "Value": {
      "Value": 9253,
      "SourceTimestamp": "2021-10-21T13:46:27.8657393Z",
      "ServerTimestamp": "2021-10-21T13:46:27.8657455Z"
    },
    "SequenceNumber": 1,
    "ExtensionFields": {
      "EndpointId": "uatf9afd7e8b1c510080c0da18e7e02a52c0b14247d",
      "PublisherId": "uatf9afd7e8b1c510080c0da18e7e02a52c0b14247d",
      "DataSetWriterId": "964e9cf9-9414-4610-9eff-a09dfdeaa337_5000"
    }
  }
```

mwiot, 2021-11-02:
Hi again, 

approx when can we expect the new container version being available in MCR, with the bug fix?

[https://mcrflowprodcentralus.data.mcr.microsoft.com/mcrprod/iot/industrial-iot-services-all](https://mcrflowprodcentralus.data.mcr.microsoft.com/mcrprod/iot/industrial-iot-services-all?P1=1635776639&P2=1&P3=1&P4=epijNVWDqXKkR7GvNxxtPCz3SZ%2BVybJlyRDmWoidLPk%3D&se=2021-11-01T14%3A23%3A59Z&sig=wqvJ0jKaqFPfc53pi1WLPKPFjMJXEwztESJ88tT%2FZEc%3D&sp=r&sr=b&sv=2015-02-21)

Best regards

cristipogacean, 2021-11-02:
Hi @mwiot,

We are in the final phase of releasing 2.8.1 version. We aim to have it available latest end of this week. 

Kind regards

cristipogacean, 2021-11-02:
Hi @teknokrat1,

Thank you very much for your feedback. This aligns very well with our strategy and plans. In the next iteration, we will be working on extending the configuration plane for the Publisher module that will cover PubSub message mode as well. 

Kind regards,
Cristian

marcschier, 2023-01-16:
2.9 release will have full OPC UA Pub Sub support, but only through direct methods and publishednodes.json, not through orchestrated mode.  Any additional work in orchestrated mode is not planned.  Closing as the 2.9 release work is tracked through other issues.

closed 2023-01-16 by marcschier