Azure/Industrial-IoT #1445: Feedback / Message / Metric on OPC Server down or wrong credentials provided in OPC Publisher 2.8+

derSchtefan, 2021-10-30:
**Is your feature request related to a problem? Please describe.**
It is currently not possible to determine whether OPC Publisher was able to connect to a server, lost the connection, received bad credentials, etc. Since we are deploying and updateing the full IoT solution from the cloud, soon to thousans of devices with 1-20 OPC servers each, we have no way of manually checking each username/password or certificate for errors. (Those are user entered in the cloud)

We are flying blind when it comes to the connection status and can often only guess with some heuristic if the OPC connection is up and running.

**Describe the solution you'd like**
OPC Publisher should report the state of each endpoint with a status code from a well-defined enum like "Connecting", "Healthy", "CouldNotConnect", "AuthenticationError". There should be no "guesswork" in error logs or codes.

The information should be made available either as reported property via the module twin, so it is available in the cloud, or via a metrics daemon like prometheus (so it is available to other modules), and should be well documented. Ideally both are available

Additionally with an opt-in via command line parameter, OPC Publsiher could  send special messages regarding connection health from the non-default edge hub output (to allow routing), or enrich messages from the default output. CouldNotConnect and AuthenticationError messages must also be sent.

Opt-in via command line argument needs to be done to not interfere with exisiting routings / processing. 

Implementing this only via edge hub messages is only acceptable if the messages of connection errors are repeated in sensible intervals, otherwise updates might get lost.

**Describe alternatives you've considered**
We have considered parsing the log output from the cloud, but this seems foolish.

cristipogacean, 2021-11-02:
@derSchtefan, thank you very much for your feedback. You're right, this is a known gap in the functionality of the publisher that we will address in the next releases.

marcschier, 2023-07-03:
We will emit connection status (EndpointConnectivityState) as OTEL metric which can be visualized using metrics collector and Grafana/AzMon.

marcschier, 2023-07-07:
As experimental feature, endpoint connectivity state is emitted as iiot_edge_publisher_client_connectivity_state gauge metric in 2.9.0. If there is feedback about how this metric should look instead, please let me know.  Or whether this info should be on the data plane (as a status message).

closed 2023-07-07 by marcschier