Azure/Industrial-IoT #1457: Debugging IoT Edge module on Visual Studio 2019

felixmondelo, 2021-11-05:
**Describe the bug**
I'm trying to debug Microsoft.Azure.IIoT.Modules.OpcUa.Publisher module on Visual Studio 2019 and it works well, but I'm not able to add CMD parameters to the module.

**To Reproduce**
Steps to reproduce the behavior:
1. Add the module to the deployment.debug.template.json with this configuration:
"Microsoft.Azure.IIoT.Modules.OpcUa.Publisher": {
            "version": "1.0.0",
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "${MODULEDIR<../modules/src/Microsoft.Azure.IIoT.Modules.OpcUa.Publisher/src>.debug}",
              "createOptions": {
                "Hostname": "OPCPublisher",
                "Cmd": [
                  "publisher",
                  "--pf=/appdata/publishednodes.json",
                  "--di=60",
                  "--to",
                  "--aa",
                  "--si=10",
                  "--ms=262144",
                  "--mm=PubSub"
                ],
                "ExposedPorts": {
                  "9702/tcp": {
                  }
                },
                "HostConfig": {
                  "Binds": [
                    "C:/appdata:/appdata"
                  ],
                  "PortBindings": {
                    "9702/tcp": [
                      {
                        "HostPort": "9701"
                      }
                    ]
                  }
                }
              }
            }
          }

2. On .\industrial-iot\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\Program.cs I don't receive any command line parameter.
3. The module is using the default publishednodes.json in the current directory instead of the parameter.

**Expected behavior**
The module should use CMD values and take the correct publishednodes.json.

**Desktop (please complete the following information):**
 - OS: Windows 10
 
**Additional context**
OPC Publisher v.2.8.1

cristipogacean, 2021-11-09:
@felixmondelo, can you detail a bit the steps you do to start the publisher module that you want to debug?

If you're using the deployment manifest and the module is started in IoT Edge context (as a docker container), you will need to attach with VS debugger after gets started. For debugging purposes there is a command line argument available in the debug build only  called `wfd` (or `waitfordebugger`). When this is used, the module will stop until the VS debugger is attached. 

If you start the module from visual studio, outside of the edge context, best would be if you could start Microsoft.Azure.IIoT.Modules.OpcUa.Publisher.Cli app and provide your arguments as arguments in it's 'launchSettings.json'

e.g.:
```{
  "profiles": {
    "Microsoft.Azure.IIoT.Modules.OpcUa.Publisher.Cli": {
        "commandName": "Project",
        "commandLineArgs": "--PkiRootPath=/mount/pki --pf=/mount/pn.json --fd=false --fm=true --bs=1 --di=20 --to --aa --si=1"
    }
  }
}
```
Please note that if you running outside of iotedge context, the deployment manifest you defined is not being applied. 

Perhaps you can find some useful information regarding edge modules debugging here: https://docs.microsoft.com/en-us/azure/iot-edge/how-to-visual-studio-develop-module?view=iotedge-2018-06

Hope this helps.

felixmondelo, 2021-11-10:
Thanks, @cristipogacean! 

I run the IoT Edge project from Visual Studio 2019, that launches the modules in the deployment manifest.

I can debug the modules, but my problem is that I can't pass in the createOptions any CMD argument because the module doesn't get them, but all the other fields are got (for example ExposedPorts or HostConfig).

cristipogacean, 2021-11-11:
@felixmondelo, would you be so kind and detail a bit for me this statement:
 "I run the IoT Edge project from Visual Studio 2019, that launches the modules in the deployment manifest."
and add some steps to reproduce.

cristipogacean, 2021-11-23:
@felixmondelo, this seem to be an issue of the IoT edge simulator not transferring properly the arguments to the module. Unfortunately we don't use this tool in our dev processes therefore no real experience with it, so we cannot really help you further. You may want to open an issue on their repo. 

Please reopen if you need support on the alternative debugging options I mentioned above.

closed 2021-11-23 by cristipogacean