Azure/Industrial-IoT #1458: China deploy.cmd minimum

Mikael159, 2021-11-05:
Hello,

I would like to deploy Azure industrial IoT on Azure China but it didn't work.

.\deploy.cmd -type minimum -version 2.7.206 -resourceGroupName myresourcegroup -subscriptionId "xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx" -aadApplicationName "Azure Industrial IoT Platform" -resourceGroupLocation chinaeast2 -localisation e2ch -environmentName AzureChinaCloud
Az installed.
AzureAD installed.
Using '2.7.206' version...
Signing in ...

To avoid logging in again next time, would you like to save your credentials? [y/n]: y
Your Azure login context will be saved into a .user file in the root of the local repo.
Make sure you do not share it and delete it when no longer needed.
Azure subscription XXXXXXXXXXX (xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx) selected.
Using existing resource group myresourcegroup ...
Deployment will use '2.7.206' branch in 'https://github.com/Azure/Industrial-IoT'.

Registering client and services AAD applications in your tenant...
Selected Tenant 'xxxxxxxx.partner.onmschina.cn' as authority.

An error occurred: Exception lors de l'appel de « Authenticate » avec « 7 » argument(s) : « Failed to acquire token silently as no token was found in the cache. Call method AcquireToken: AADSTS500011: The resource principal named https://graph.microsoft.com was not found in the tenant named XXXXXXXXXX. This can happen if the application has not been installed by the administrator of the tenant or consented to by any user in the tenant. You might have sent your authentication request to the wrong tenant.
Trace ID: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
Correlation ID: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
Timestamp: 2021-11-05 10:36:51Z »

Ensure you have installed the AzureAD cmdlets:
1) Run Powershell as an administrator
2) in the PowerShell window, type: Install-Module AzureAD

Exception lors de l'appel de «Authenticate» avec «7» argument(s): «Failed to acquire token silently as no token was found in the cache.
Call method AcquireToken: AADSTS500011: The resource principal named https://graph.microsoft.com was not found in the tenant named
XXXXXXXXXX. This can happen if the application has not been installed by the administrator of the tenant or consented to by any user in
the tenant. You might have sent your authentication request to the wrong tenant.
Trace ID: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
Correlation ID: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
Timestamp: 2021-11-05 10:36:51Z »
Au caractère XXXXXXXXXXXXX\Azure-Industrial-IoT-2.7.206\deploy\scripts\aad-register.ps1:496 : 9
+         $token = [Microsoft.Azure.Commands.Common.Authentication.Azur ...
+         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : AadAuthenticationFailedException

Could you help me ?
Thanks

marcschier, 2021-11-05:
Only Azure Global cloud is currently supported.  I have tagged this as a feature request.

Mikael159, 2021-11-05:
Hello,
Thank you for your answer.
Is there a workaround?
best regards

marcschier, 2021-11-06:
You should be able to make the changes to select a different cloud and related endpoints in the PowerShell scripts and templates in the deploy folder.  We have no detailed instructions what changes those are though.

Mikael159, 2021-11-08:
Hello,
I have already modified it.
AzureCloud by AzureChinaCloud
on these files : 
- deploy/scripts/deploy.ps1
- deploy/scripts/aad-register.ps1
What else do I need to do ?
Thanks

hansgschossmann, 2021-11-09:
@Mikael159  you need to address the issues one by one. Do you still see the same issue as in your first post?

marcschier, 2021-11-09:
Hi @Mikael159, the service endpoints for all Azure services in China cloud are different than in Global cloud, they need to be provided to the templates, and ultimately services.  Note that only Global cloud is tested, and thus supported at this point (i.e. what can be deployed with the deployment scripts provided).  We have tagged this as feature request and will evaluate it for inclusion into a future release.

Mikael159, 2021-11-10:
Hello,

I have changed all "https://graph.microsoft.com" by "https://microsoftgraph.chinacloudapi.cn"

The deployement start but crash at keyvault creation : 
Starting deployment...
10:49:11 - The deployment 'azuredeploy' failed with error(s). Showing 2 out of 2 error(s).
Status Message: Premium vaults are not supported in this region. (Code:SkuNotSupported)

Status Message: The resource operation completed with terminal provisioning state 'Failed'. (Code: ResourceDeploymentFailure)
 - At least one resource deployment operation failed. Please list deployment operations for details. Please see https://aka.ms/DeployOperations for usage details. (Code: DeploymentFailed)
   - {
  "error": {
    "code": "SkuNotSupported",
    "message": "Premium vaults are not supported in this region."
  }
} (Code:BadRequest)
CorrelationId: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
Deployment failed.

I have changed in deploy/templates/azuredeploy.minimum.json on keyVaultSkuName section "Premium" by "Standard"
But this is the same error.

Where can i force Standard Sku to keyvault creation ?

Thanks

chmagitt, 2021-11-17:
Hello

We are stuck in the deployment of I-IoT in e2ch region
the deployment creates the following services : Cosmos, EH, IoT hub, Managed Identity, Route Table, Storage Account, Service Bus
then it breaks when it comes to Keyvault

[image]

chmagitt, 2021-11-17:
Hello

back to Azure portal we've notice this in the failed deployment logs 

{
  "id": "/subscriptions/85dxxxxxxxxxxxxb3d/resourceGroups/gsebdsppdaiie2chrsg00/providers/Microsoft.Resources/deployments/infrastructure.minimum",
  "name": "infrastructure.minimum",
  "type": "Microsoft.Resources/deployments",
  "properties": {
    "templateLink": {
      "uri": "https://raw.githubusercontent.com/Azure/Industrial-IoT/2.7.206/deploy/templates/azuredeploy.minimum.json",
      "contentVersion": "1.0.0.0"
    },

and this  azuredeploy.minimum.json  has Key Vault configured with Premium SKU

hansgschossmann, 2021-12-22:
We have discussed this in more detail and we are not going to support deployment in china because of various required resources are not available in china. Closing it for now.

closed 2022-01-12 by hansgschossmann