Azure/Industrial-IoT #1465: OPC publisher & Azure Cosmos DB RU consumption

b-enigma-con, 2021-11-12:
L.S.,

We've deployed the IIoT stack in Azure, using the microservices setup (see https://azure.github.io/Industrial-IoT/services/). We currently have one IoT Edge device running the following modules (amongst others): OPC twin and OPC publisher. To start with, we configured an initial limit of 400 RU’s for the “iiot_opc” container in the Azure Cosmos DB account.

After adding approx. 3500 - 4000 published nodes, using the microservices OPC publisher REST API, we discovered a lot of 429 failures in Cosmos DB (meaning that requests are being throttled because of the RU limit). We increased the limit to 600 RU’s and added up to approx. 6600 published nodes, and again our RU limits were hit for the “iiot_opc” container.

After turning on Cosmos DB diagnostic logs to inspect the incoming requests, I noticed that sometimes we receive up to 10 Cosmos DB update requests per second with the “RequestResourceID” pointing to our IoT Edge device.

Is this by design (and if so, why is this)? Are we supposed to keep on increasing the RU limit for the “iiot_opc” container each time we add new published nodes? This might become quite expensive in the future.

Thank you!

cristipogacean, 2021-11-18:
@b-enigma-con, thank you for bringing this up. 

A workaround for this issue is documented in the [Operations Manual ](https://github.com/Azure/Industrial-IoT/blob/main/docs/manual/readme.md)

> If the request duration is too high, that can mean that the service needs more resources than are currently assigned.
> 
> Steps to verify that:
> 
> - check the memory and CPU consumption on the pod where the service runs to see if they are too high. More than 1 GB of memory consumption and more than 250 millicores of CPU consumption would be considered high.
> 
> - check the CosmosDB instance in the Azure portal for alerts like "number of requests exceeding capacity".
> 
> If you discover one of these issues, it’s either possible to reduce the resource consumption of the service or to increase assigned resources:
> 
> 1. Reduce resource consumption by adjusting heartbeat and job checking intervals
> 
>    Increasing of the heartbeat and the job check intervals (p.e. to 2 minutes) reduces the overall number of requests and therefore results in a lower usage of resources. However, with increasing interval changes on job will take longer to get reflected by the publisher. To change the intervals, the Registry-Service (see 2.2.1) can be used like this:
> 
>    - Get a list of running publishers: `GET /registry/v2/publishers`
> 
>      The response will contain a list with active publishers, each of them has a property Configuration with sub-properties `JobCheckInterval` and `HeartbeatInterval`
> 
>    - Update the publisher configuration with the updated `JobCheckInterval` and `HeartbeatInterval`-Properties: `PATCH /registry/v2/publishers/<PublisherId>`
> 
>      The details of these operations can be found by using the Swagger-Definition of the Registry-Service: `/registry/swagger/`
>
...

hansgschossmann, 2021-11-23:
@b-enigma-con has this workaround fixed your problem?

b-enigma-con, 2021-12-02:
I will have a check next week, thank you for your advice!

b-enigma-con, 2021-12-07:
We increased the JobCheckInterval and HeartbeatInterval to 2 hours, but the result is not what we hoped for. Attached image is our QA environment, the changes are applied at the yellow mark. 

10.000 Cosmos DB RU is not enough in our production environment with +/- 20500 published nodes, causing 429 responses.

[image]

We're wondering what is causing so many update requests to the Cosmos DB. Are we reaching the limits of the IIoT Microservices architecture with this amount of nodes?

cristipogacean, 2021-12-10:
@b-enigma-con, thank you for the investigation! It looks like you have run into a bug/regression that prevents the publisher module to apply the newly configured  Heartbeat and JobCheck intervals during runtime. A workaround for it is to manually restart the publisher modules, then the configured values are being picked-up. Furthermore, for heartbeat value the maximum configurable value is 1 minute, which will be a significant improvement compared to the default of 3 seconds you have noticed. 

you can restart the publisher module directly from Azure portal following the steps: 
1. Open the IoTEdge device from the IotHub resource.
2. Find the publisher module and open the troubleshoot page follow the "Running" link as in the picture below.
[image]
3. Hit "Restart Publisher" button in the top right corner.
[image]

This shall apply the new intervals the publisher module is asking the cloud based publisher orchestrator microservice for configuration changes. 

I will mark this issue as bug and will get fixed in the next release.

closed 2023-05-19 by marcschier