Azure/Industrial-IoT #1469: pn.json schema validation backwards compatibility issues

cristipogacean, 2021-11-16:
@WilliamBerryiii, @suneetnangia, @karok2m, setting up an issue to continue the conversation within the PR https://github.com/Azure/Industrial-IoT/pull/1464.

@WilliamBerryiii, we would really appreciate your help as a PR here. Your plan looks perfect to tackle this topic. I totally agree that mixing (Node)Id and ExpandedNodeId usage in the same configuration is a bit odd, especially since the Id is so flexible and can accommodate all possible node id variants. On the other hand I know that when people have (many) options they get creative for various reasons (e.g. incomplete documentation). Furthermore, in the past, our users did not experience the smoothest version upgrades, especially for the publisher. Now we need to make sure we're not breaking backyards compatibility and if we do so we must have good reasons and explain them to our users.

As for your final notes, I really appreciate them and I would love to have a call for us to share our plans and ideas and identify the points where we can support each-other.

WilliamBerryiii, 2021-11-17:
@cristipogacean - sounds good!  

We will draft a PR with the following supports:

- [ ] Update the Schema Generator tool to produce a schema that only performs type validation on `ID` and `ExpandedNodeId` fields, removing value validation.
- [ ] Update the Schema Generator tool to produce a schema that allows both `ID` and `ExpandedNodeId` to be present in an `OpcNodes` array item.
- [ ] Update the Schema Generator tool's unit tests to cover the pn.json sample provided in PR #1464 (will not be attached to this repo; but we'll link back for thread consistency).
- [ ] Add unit tests to OPC UA Schema Validation tests in the OPC Publisher project to cover the pn.json sample provided in PR #1464 
- [ ] Regenerate a new schema with above modifications and overwrite the version in the OPC Publisher project
- [ ] Update schema spec doc included in the OPC Publisher project to include references to this conversation and adjustments made to the schema

I think that should be the full working list to close this issue; let me know if there are other provisions/changes that should be made. 

cc/ @suneetnangia

WilliamBerryiii, 2021-11-17:
@cristipogacean as I'm looking at the following sample and finding myself confused as to how OpcNodes with both `ID` and `ExpandedNodeId` defined are supposed to behave.

```
"DisplayName": "Kep_2_DisplayName",
"DataSetFieldId": "Kep_2_DataSetFieldId",
"Id": "ns=2;s=Sim.CH1.SIM_CH1_TAG10\\2347798-c63-19401",				"ExpandedNodeId":"nsu=KEPServerEX;s=Sim.CH1.SIM_CH1_TAG10\\2347798-c63-19401",
"OpcSamplingInterval": 1000,
"OpcPublishingInterval": 1000
```

I can reconcile that these two values "could" be present on the same item; but it seems like this should at minimum yield a warning, where users are told which identifier OPC Publisher will actually use to access a given node. With `Id` defined first is that the utilized value? If `Id` were second would is use the `ExpandedNodeId`? I.E. should we create a tracking issue where this condition, though supported by the serializer, gives a detailed warning message about how precedent is handled?

cristipogacean, 2021-11-18:
@WilliamBerryiii, right, in the actual implementation, when `Id` is present the `ExpandedNodeId` is simply ignored. When `Id` is not present the `ExpandedNodeId` is interpreted as if it was `Id` and processed accordingly.
I like the idea of issuing a warning when `ExpandedNodeId`s are found. Perhaps a "ExpandedNodeId is a legacy attribute (use Id instead)" like warning?

closed 2022-04-01 by cristipogacean