Azure/Industrial-IoT #1473: Authenticate to OPC UA Server with username and password

ahe27, 2021-11-19:
I can discover my OPC Server using the Engineering Tool. 
While browsing the adress space of the endpoint I am not able to discover the nodes of my iot device. I guess this is since the device is secured with username and password. As far as i know i cannot authenticate to the UA Server with credentials using the Engineering Tool.

Is there an other way to subscribe to the values of those nodes?

hansgschossmann, 2021-11-23:
@ahe27 can you clarify your issue description. you are stating that your are browsing the address space of the endpoint. i assume that you have been able to connect to the endpoint successfully. connecting to the endpoint is typically where you provide your username/password. if you can connect and browse, then you will see the nodes which the system administrator of the OPC UA server has configured for you to see. there is not workaround for this security configuration.
so pls check with the system integrator of your OPC UA server.

ahe27, 2021-11-24:
My issue has been resolved.

I can explore my Server even when I am not authenticated but several nodes are hidden this way.
I overlooked the "Asset Login" section. After entering the credentials of the server in this section I can see the corresponding nodes.
At first it was not clear, that those credentials are stored for the session and used for all following server requests.

closed 2021-11-24 by ahe27