Azure/Industrial-IoT #1476: PublishedNodesFile encoding

felixmondelo, 2021-11-22:
**Describe the bug**
Our clients have the OPC signals with symbols or accents like:

 "OpcNodes": [
      {
        "Id": "ns=2;s=_AdvancedTags.PLC.Producción.PLC1",
        "OpcSamplingInterval": 1000,
        "OpcPublishingInterval": 1000,
        "HeartbeatInterval": 3000,
        "SkipFirst": false
      },
      {
        "Id": "ns=2;s=_AdvancedTags.PLC.Temperature (ºC)",
        "OpcSamplingInterval": 1000,
        "OpcPublishingInterval": 1000,
        "HeartbeatInterval": 3000,
        "SkipFirst": false
      }

The problem is that when the OPC module executes and read the publishednodes.json, It throws an exception:

Microsoft.Azure.IIoT.Exceptions.SerializerException: Validation failed with error: Expected 1 matching subschema but found 0 at schema path: #/items/oneOf, and configuration file location #/0; Validation failed with error: The string value was not a match for the indicated regular expression at schema path: #/items/oneOf/0/properties/OpcNodes/items/properties/Id/pattern, and configuration file location #/0/OpcNodes/0/Id; Validation failed with error: Required properties [ExpandedNodeId] were not present at schema path: #/items/oneOf/1/properties/OpcNodes/items/required, and configuration file location #/0/OpcNodes/0; Validation failed with error: Required properties [NodeId] were not present at schema path: #/items/oneOf/2/required, and configuration file location #/0

**To Reproduce**
Steps to reproduce the behavior:
Deploy OPC Publisher with a publishednodes.json configuration file with accents or symbols.

**Expected behavior**
The module read the publishednodes.json file and continues processing the OPC signals.

felixmondelo, 2021-11-22:
Seems that is the publishednodes schema file and not the encoding ...

cristipogacean, 2021-11-22:
@felixmondelo, indeed the issue seem to be caused by the schema validation. Which version of the publisher do you use? If possible please use the most recent release - 2.8.1 (where the schema validation is not enabled) or update your main branch. The schema validation was temporarily disabled in main branch with the PR: https://github.com/Azure/Industrial-IoT/pull/1464.

felixmondelo, 2021-11-23:
I have the version 2.8.1-rc.43+d172d55062, the problem solves when add the parameter pfs to empty.

cristipogacean, 2021-12-03:
Closing since the most recent released version does not have this issue and the issue was fixed on main branch in the meanwhile.
Please reopen if you need more support.

closed 2021-12-03 by cristipogacean