Azure/Industrial-IoT #1477: OPC Publisher > Is mandatory encription when UseSecurity OpcAuthenticationMode=UsernamePassword

felixmondelo, 2021-11-23:
**Describe the bug**
If we want use UseSecurity = true and OpcAuthenticationMode = UsernamePassword, the module tries to connect with encryption (SecurityMode.Best).

In Industrial-IoT\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Edge.Publisher\src\Storage\PublishedNodesJobConverter.cs, at ToWriterGroupJobs sets this condition:

SecurityMode = item.UseSecurity == false && item.OpcAuthenticationMode != OpcAuthenticationMode.UsernamePassword ? SecurityMode.None : SecurityMode.Best

Is possible to configure don't use SecurityMode.Best and encryption when security is enabled with UsernamePassword?

hansgschossmann, 2021-11-23:
@felixmondelo which SecurityMode do you want to select instead of Best. The architecture of the Industrial-IoT platform is to use the highest possible security which is supported by client (OPC Publisher) and server (OPC UA server).

felixmondelo, 2021-11-23:
@hansgschossmann I know that is the best option, but we have customers that use UsernamePassword without encryption because the connection is behind a VPN. I need that this can be configurable.

hansgschossmann, 2021-12-02:
@felixmondelo thanks for your answer. i will follow up with the dev team on this.

felixmondelo, 2021-12-02:
I think that the modification is on Industrial-IoT\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Edge.Publisher\src\Storage\PublishedNodesJobConverter.cs class on the method ToWriterGroupJobs (in bold the conflicting line):

```
private IEnumerable<WriterGroupJobModel> ToWriterGroupJobs(
            IEnumerable<PublishedNodesEntryModel> items,
            LegacyCliModel legacyCliModel) {
            if (items == null) {
                return Enumerable.Empty<WriterGroupJobModel>();
            }
            try {
                // note: do not remove 'unnecessary' .ToList(),
                // the grouping of operations improves perf by 30%
                // Group by connection
                var group = items.GroupBy(
                    item => new ConnectionModel {
                        OperationTimeout = legacyCliModel.OperationTimeout,
                        Id = item.DataSetWriterId,
                        Group = item.DataSetWriterGroup,
                        Endpoint = new EndpointModel {
                            Url = item.EndpointUrl.OriginalString,
                            SecurityMode = item.UseSecurity == false &&
                            **item.OpcAuthenticationMode != OpcAuthenticationMode.UsernamePassword ?
                            SecurityMode.None : SecurityMode.Best**
                        },
                        User = item.OpcAuthenticationMode != OpcAuthenticationMode.UsernamePassword ?
                        null : ToUserNamePasswordCredentialAsync(item).Result,

                    },
```

Maybe here instead of "SecurityMode.None : SecurityMode.Best" should be a configuration parameter.

felixmondelo, 2022-01-21:
Hi @hansgschossmann how do you consider progress with this?

cristipogacean, 2022-02-09:
The Security mode should be independent on the user authentication mode.

Fix is  in https://github.com/Azure/Industrial-IoT/pull/1494. The fix will be part of version 2.8.2

felixmondelo, 2022-03-25:
@cristipogacean I'm trying the new 2.8.2 version, but I don't know how this issue is supposed to be fixed, can you explain?