Azure/Industrial-IoT #1481: OPC Twin > Only one OPC UA Server

felixmondelo, 2021-11-26:
**Describe the bug**
Hi, I've got only one OPC Server UA and one IoT Edge device configured with OPC Publisher. I want to use OPC Twin module, but there is missing documentation in https://github.com/Azure/Industrial-IoT/blob/main/docs/architecture-flow.md.

How is the simple way to configure this scenario? How are the steps to best configure this scenario?

hansgschossmann, 2021-12-02:
@felixmondelo I assume you have the cloud microservices deployed, since using the modules standalone is only supported for OPC Publisher, but not for OPC Twin.
You may also want  to check the operations manual for some details: https://azure.github.io/Industrial-IoT/manual/

cristipogacean, 2021-12-03:
@felixmondelo, indeed, before making any browse/read/write, etc operations with the twin you need to discover the endpoint and then activate it. Activation operation instructs the twin to create and activate a session into the OPC UA Server so the other service calls are possible.

jehona-m, 2021-12-07:
Please reactivate if further help is needed.

closed 2021-12-07 by jehona-m