Azure/Industrial-IoT #1483: OPC-Twin-Service v2 Swagger's "Try it out" it's working?

felixmondelo, 2021-11-29:
**Describe the bug**
I want to test the OPC Twin Service v2 through "Try it out" utility from Swagger (http://localhost:9041/swagger/index.html for example).

When I call any of the OPC Twin Service v2 I always get the same exception:

`"ResponsePayload": "{\"ClassName\":\"System.ArgumentNullException\",\"Message\":\"Value cannot be null.\",\"Data\":null,\"InnerException\":null,\"HelpURL\":null,\"StackTraceString\":\"   at Microsoft.Azure.IIoT.Modules.OpcUa.Twin.Controllers.SupervisorMethodsController.BrowseAsync(EndpointApiModel endpoint, BrowseRequestInternalApiModel request) in C:\\\\Industrial-IoT\\\\modules\\\\src\\\\Microsoft.Azure.IIoT.Modules.OpcUa.Twin\\\\src\\\\Controllers\\\\SupervisorMethodsController.cs:line 81\",\"RemoteStackTraceString\":null,\"RemoteStackIndex\":0,\"ExceptionMethod\":null,\"HResult\":-2147467261,\"Source\":\"Microsoft.Azure.IIoT.Modules.OpcUa.Twin\",\"WatsonBuckets\":null,\"ParamName\":\"request\",\"Exception\":\"ArgumentNullException\"}"`

Investigating, the problem is that in the OPC Twin module, I'm not receiving a request with endpoint and request parameters in the json payload, for example:

```
{
    "methodName": "Browse_V2",
    "responseTimeoutInSeconds": 200,
    "payload": {
    	"endpoint": {
    		"Url": "opc.tcp://10.10.10.10:44444",
    		"SecurityMode": "None"
    	},
    	"request": {
    			"NodeIdsOnly": true
		}
    }
}
```
I receive instead this payload

```
{
	"NodeId": null,
	"Direction": null,
	"View": null,
	"ReferenceTypeId": null,
	"NoSubtypes": null,
	"MaxReferencesToReturn": null,
	"TargetNodesOnly": true,
	"ReadVariableValues": true,
	"NodeIdsOnly": null,
	"NodeClassFilter": null,
	"Header": null
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Start the OPC Twin Services v2 service.
2. Go to the Swagger (http://localhost:9041/swagger/index.html)
3. Press "Try it out" in any of the methods.
4. Press execute.

**Expected behavior**
Receive a correct payload in the Twin module, to process the call.

hansgschossmann, 2021-12-02:
@felixmondelo have you validated that the Engineering Tool is able to access and browse your OPC UA server?

felixmondelo, 2021-12-02:
Hi @hansgschossmann, I can access directly through OPC Twin's Direct Methods, for example:

```
{
    "methodName": "ValueRead_V2",
    "responseTimeoutInSeconds": 200,
    "payload": {
    	"endpoint": {
    		"Url": "opc.tcp://10.10.10.10:49999",
    		"SecurityMode": "None"
    	},
    	"request": {
    		"Header": {
    			"Elevation": {
    				"Type": "UserName",
    				"Value": {
    					"user": "user1",
    					"password": "password1"
    				}
    			}
    		},
			"NodeId": "ns=2;s=E_Aleton_Moda.Mino1.Speed"
		}
    }
}
```
And receive the response from OPC:

```
{
    "status": 200,
    "payload": {
        "value": 2.811072,
        "dataType": "Float",
        "sourceTimestamp": "2021-11-30T10:28:46.9645891Z",
        "serverTimestamp": "2021-11-30T10:28:46.9645891Z"
    }
}
```

marcschier, 2021-12-03:
The REST calls require a endpoint id, not endpoint object.  Is it possible that this endpoint id does not exist?  It is the twin identity that is created after a discovery run completes and finds the endpoint.

felixmondelo, 2021-12-03:
@marcschier I'm talking about the request received by the Twin module. 

The request reaches the correct endpoint because from Swagger I add the Endpoint Id parameter and I received it on my Twin module in the IoT Edge.

The request reaches zipped and the content it's:

```
{
	"NodeId": null,
	"Direction": null,
	"View": null,
	"ReferenceTypeId": null,
	"NoSubtypes": null,
	"MaxReferencesToReturn": null,
	"TargetNodesOnly": true,
	"ReadVariableValues": true,
	"NodeIdsOnly": null,
	"NodeClassFilter": null,
	"Header": null
}
```

Instead of something similar to this:

```
{
    "methodName": "Browse_V2",
    "responseTimeoutInSeconds": 200,
    "payload": {
    	"endpoint": {
    		"Url": "opc.tcp://10.10.10.10:44444",
    		"SecurityMode": "None"
    	},
    	"request": {
    			"NodeIdsOnly": true
		}
    }
}
```

karok2m, 2021-12-10:
@felixmondelo, can I please ask you to provide a screenshot from the Swagger UI with your input parameters and with the result after you hit `Execute` button ?

hansgschossmann, 2021-12-21:
@felixmondelo have you been able to solve your issue meanwhile?

felixmondelo, 2021-12-22:
@karok2m this are the screenshots:

[image]

[image]

@hansgschossmann the problem continues, the body that the module OPCTwin on the IoT Edge receives is incorrect from the Swagger as you can see in the previous screenshot.

hansgschossmann, 2022-01-11:
@karok2m can you provide a sample on how to call the /v2/browse API via swagger. I am not sure if we have an example in our documentation.

karok2m, 2022-01-18:
@felixmondelo, `endpointId` that you are providing does not look correct. It should be a GUID taken from Registry microservice, there is a `Endpoints` section for endpoints-related operations. One of them is getting all available endpoints. Please note that you need an activated endpoint to browse.

For example this is what I would have to do to browse an endpoint for a new server:

1. Discover a server: for this I am using Swagger of Registry microservice, specifically `/registry/v2/applications/discover` endpoint:

[image]

Here I am trying to discover endpoints of a server at `opc.tcp://20.76.63.47:50000/` address.

2. Get available endpoints: for this I am again using Swagger of Registry microservice, specifically `/registry/v2/endpoints` endpoint. This will show me the endpoints that the platform knows about. For my server it discovered two endpoints. At first all of the endpoints will have `Deactivated` state. We will activate the first one (with id `uatb998698cdb686e4a5e9c08bc80f30081ecb79308`) in the next step.

[image]

3. Activate an endpoint that I want to browse: for this I am again using Swagger of Registry microservice, this time `/registry/v2/endpoints/{endpointId}/activate` endpoint. The value that I use here as `endpointId` is `uatb998698cdb686e4a5e9c08bc80f30081ecb79308`.

[image]

Now if you get list of all endpoints again, the endpoint with `uatb998698cdb686e4a5e9c08bc80f30081ecb79308` id should have `activationState` set to `ActivatedAndConnected` and `endpointState` set to `Ready`. This means that OPC Twin module was able to establish a connection to this endpoint of my server. Only after this I will be able to browse it.

4. Browse the endpoint: now I will use `/twin/v2/browse/{endpointId}` endpoint of Twin microservice using its Swagger. Please note that here again, I am using `uatb998698cdb686e4a5e9c08bc80f30081ecb79308` as `endpointId`:

[image]

As you can see in the screenshot, I am able to get top level nodes of the endpoint. You would need to browse each node in similar fashion to see its child nodes.

@felixmondelo, can you please confirm that you are able to browse the nodes using these steps ?

hansgschossmann, 2022-01-25:
@felixmondelo are you able to browse nodes using the steps @karok2m has  provided?

felixmondelo, 2022-01-25:
Yes, sorry for the delay and thanks to @karok2m.

hansgschossmann, 2022-02-01:
Thanks @karok2m for your comprehensive answer.

closed 2022-02-01 by hansgschossmann