Azure/Industrial-IoT #1490: Cosmos DB document limit reached, what to do?

b-enigma-con, 2021-12-06:
L.S.,

We would like some guidance on the following issue. We've deployed the IIoT stack in Azure, using the microservices setup (see https://azure.github.io/Industrial-IoT/services/). We currently have one IoT Edge device running the following modules (amongst others): OPC twin and OPC publisher. We have quite some published/monitored nodes, and our Cosmos DB document (storing our published node configuration) has reached its limit (2MB). We're planning on adding at least twice the amount of published nodes to this system, but we're stuck on how to proceed. What is the recommended way to add more published nodes in this case?

Thank you!

b-enigma-con, 2021-12-07:
As can be read here (https://azure.github.io/Industrial-IoT/manual/#limits-and-contributing-factors):

_**Size of job definitions**
Job definitions that are stored in Cosmos DB have 2MB limit on their size. This puts an indirect limit on number of nodes that can be put in a single job._

I can't find any documentation on how to handle this situation.

hansgschossmann, 2021-12-07:
@b-enigma-con the 2MB are a limitation of Azure Cosmos DB. The only way to address this is to change how we store the nodes in the document, which is a feature request.
I am curious to know how many nodes you are trying to publish when you hit the 2MB limit.

b-enigma-con, 2021-12-07:
Currently about 20500 nodes, and we are planning on using +/- 75000 nodes in the very near future. This is the same setup as mentioned in issue #1465.

closed 2023-05-19 by marcschier