Azure/Industrial-IoT #1492: Unhandled error during Publish.  BadTimeout 'BadTimeout'

mwiot, 2021-12-07:
Hi, we are having some issues with the publisher. Getting bad timeout. Can you see from the logs what is causing this?
Had a stream running for the publisher for several hours, that suddenly stopped working. When checking the logs there were errors there regarding bad timeout. After the restart it is publishing data, but this error keeps coming on regular intervals. This is some timeout error related to OPC client (publisher) and OPC UA server?

```
[13:46:37 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) PUBLISH #15 - Unhandled error during Publish.
 BadTimeout 'BadTimeout'
```

Running 
Ubuntu 18.04
IIOT 2.8.1 all in one services
Publisher 2.8.1

```
[13:46:36 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Sending heartbeat...
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] Sending POST request to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat...
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat returned OK (took 00:00:00.0518594).
[13:46:36 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker+JobProcess] Sending job processor heartbeat...
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] Sending POST request to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat...
[13:46:36 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Runner Keepalive reset due to checkAlive
[13:46:36 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Refreshing session '2cc38d59fc2d08bc4bf6abe37aecde7d3ffe0640'
[13:46:36 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Service) Read Called. RequestHandle=394, PendingRequestCount=2
[13:46:36 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Service) Read Completed. RequestHandle=394, PendingRequestCount=1
[13:46:36 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Refreshing done for session '2cc38d59fc2d08bc4bf6abe37aecde7d3ffe0640'
[13:46:36 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Sending heartbeat...
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] Sending POST request to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat...
[13:46:36 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Sending heartbeat...
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] Sending POST request to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat...
[13:46:36 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Sending heartbeat...
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] Sending POST request to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat...
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat returned OK (took 00:00:00.0887432).
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat returned OK (took 00:00:00.0477676).
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat returned OK (took 00:00:00.0484443).
[13:46:36 VRB Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://svc-bifrost-p2-weu.azurewebsites.net/edge/publisher/v2/heartbeat returned OK (took 00:00:00.0529546).
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) PUBLISH #15 RECEIVED
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Service) ***EXCEPTION*** BadTimeout
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Service) Publish Completed. RequestHandle=0, PendingRequestCount=0, StatusCode=Bad
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) Publish #15, Reconnecting=False, Error: BadTimeout
[13:46:37 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) PUBLISH #15 - Unhandled error during Publish.
 BadTimeout 'BadTimeout'

========================================
Id: BadTimeout
Description: BadTimeout
>>> BadTimeout
---    at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
---    at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
---    at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
========================================

Opc.Ua.ServiceResultException: BadTimeout
   at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
   at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Service) Publish Called. RequestHandle=16, PendingRequestCount=1
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) PUBLISH #16 SENT
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) PUBLISH #16 RECEIVED
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Service) Publish Completed. RequestHandle=16, PendingRequestCount=0
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) Error - Publish call finished. ResultCode=BadSequenceNumberUnknown; SubscriptionId=2403595714;
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) NOTIFICATION RECEIVED: SubId=2403595714, SeqNo=8
[13:46:37 DBG Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Keepalive received from session '2cc38d59fc2d08bc4bf6abe37aecde7d3ffe0640': server current state: Running
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) NotificationReceived: ClientHandle=169, Value=8
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) NotificationReceived: ClientHandle=94, Value=3.2
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) NotificationReceived: ClientHandle=133, Value=6.5
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) NotificationReceived: ClientHandle=129, Value=3.2
[13:46:37 VRB Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Information) NotificationReceived: ClientHandle=49, Value=0
```

Best regards

cristipogacean, 2021-12-07:
This seem like either a networking issue between the publisher and the OPC UA Server, or the server is becoming unresponsive. Do you have a chance to look into the logs of the server and check if something goes wrong there?

mwiot, 2021-12-07:
Hi, yes. 
What is the default settings for the container create option "ot|operationtimeout" "The operation timeout of the publisher OPC UA client in ms." ?
Can it help to alter the timeout?

Did not find anything in the logs. The UA server is a DA/UA converter from Softing.

cristipogacean, 2021-12-07:
Default operation timeout is 15seconds. I'm not sure though it it makes sense to increase it. Especially when the publishing interval is much smaller than 15seconds.  Perhaps it makes more sense to increase the publishing interval when this is acceptable.

mwiot, 2021-12-07:
Currently we are sending every value every 2 minutes even if there are no value changes and are using these settings per for every nodeid. 215 tags configured like this:

      samplingInterval: 00.00:02:00
      publishingInterval: 00.00:02:00
      HeartbeatInterval: 00.00:02:00

cristipogacean, 2021-12-07:
oh, in this case let's try to increase the operation timeout.

mwiot, 2021-12-08:
Tried to set operation time to 60 000msec, but the error  still shows up. Any other ideas?

cristipogacean, 2021-12-10:
We have seen similar issues when the the host machine running the IoTEdge runtime/publisher is under heavy load. Could it be the case here?

mwiot, 2021-12-10:
The stream has now been running for 2 days continuously, but the error still shows up in the logs.

mwiot, 2021-12-10:
The load seems fine?

[image]

cristipogacean, 2021-12-10:
Indeed, that looks fine.

You mentioned you see the exception on regular basis, how often does this timeout occur? is it somehow in the same cycle as the operation timeout?
Would you be able to provide us a larger log snippet that contains multiple exceptions so we can further investigate?

Do you see any data loss when the exception occurs?
I will mark this thread as a bug now since we need to understand better the cause of this timeout.

mwiot, 2021-12-10:
```

[12:20:12 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) PUBLISH #59 - Unhandled error during Publish.
 BadTimeout 'BadTimeout'

Opc.Ua.ServiceResultException: BadTimeout
   at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
   at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
[12:20:12 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : uat812748106293fcf387e1c4922cb0e7bbf1b8c049
  # Ingestion duration                 :    00:01:04:00 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :             32 (All time ~0.01/s; 00 in last 60s ~0/s)
  # Ingress ValueChanges (from OPC)    :          6,355 (All time ~1.65/s; 00 in last 60s ~0/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :          6,355
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :             32
  # Encoder avg Notifications/Message  :            199
  # Encoder avg IoT Message body size  :         24,762 (9%)
  # Encoder avg IoT Chunk (4 KB) usage :              6
  # Estimated IoT Chunks (4 KB) per day:          5,040
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :             32 (0.01/s)
  # Connection retries                 :              0

[12:20:33 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 1: Module kkv-edge-gateway_publisher disconnected due to Communication_Error - now Disconnected_Retrying...
[12:20:33 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 1: Device DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049 disconnected due to Communication_Error - now Disconnected_Retrying...
[12:20:33 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 1: Module kkv-edge-gateway_publisher reconnected due to Connection_Ok.
[12:20:33 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 1: Device DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049 reconnected due to Connection_Ok.
[12:21:12 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : uat812748106293fcf387e1c4922cb0e7bbf1b8c049
  # Ingestion duration                 :    00:01:04:59 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :             33 (All time ~0.01/s; 01 in last 60s ~0.02/s)
  # Ingress ValueChanges (from OPC)    :          6,570 (All time ~1.68/s; 215 in last 60s ~3.58/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :          6,570
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :             33
  # Encoder avg Notifications/Message  :            199
  # Encoder avg IoT Message body size  :         24,823 (9%)
  # Encoder avg IoT Chunk (4 KB) usage :            6.1
  # Estimated IoT Chunks (4 KB) per day:          5,118
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :             33 (0.01/s)
  # Connection retries                 :              0

[12:21:25 INF Microsoft.Azure.IIoT.Modules.OpcUa.Publisher.Controller.ConfigurationSettingsController] Agent configuration updated.
[12:21:25 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] New settings processed.
[12:22:12 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : uat812748106293fcf387e1c4922cb0e7bbf1b8c049
  # Ingestion duration                 :    00:01:06:00 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :             33 (All time ~0.01/s; 00 in last 60s ~0/s)
  # Ingress ValueChanges (from OPC)    :          6,570 (All time ~1.66/s; 00 in last 60s ~0/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :          6,570
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :             33
  # Encoder avg Notifications/Message  :            199
  # Encoder avg IoT Message body size  :         24,823 (9%)
  # Encoder avg IoT Chunk (4 KB) usage :            6.1
  # Estimated IoT Chunks (4 KB) per day:          5,040
  # Outgress Batch Block buffer size   :              0
[... 609 lines cut ...]
Opc.Ua.ServiceResultException: BadTimeout
   at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
   at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
[... cut ...]
  # Encoder avg Notifications/Message  :            196
  # Encoder avg IoT Message body size  :         24,493 (9%)
  # Encoder avg IoT Chunk (4 KB) usage :              6
  # Estimated IoT Chunks (4 KB) per day:          4,320
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :             46 (0.01/s)
  # Connection retries                 :              0

```

cristipogacean, 2021-12-10:
I managed to replicate this issue in my environment. On a first look it seems like a miss-configuration issue between the publishing interval and the keep-alive count on the publisher side but the issue requires deeper investigation.

The good news is I don't see any loss of data in the upstream. I'm assuming the publish request that expires is empty and the next one that succeeds will return the queued values within the server anyhow. This will sightly increase the communication between the publisher & opc ua server, but should not affect the overall communication behavior. It is uncompliant behavior though so will be fixed in the next release.

Thank you very much for supporting us on this matter!

mwiot, 2021-12-20:
Thanks for putting fix in the next release.

cristipogacean, 2022-04-01:
Fixed in 2.8.2.

closed 2022-04-01 by cristipogacean

aleducq, 2022-08-12:
Hello @cristipogacean , 
I'm facing a similar issue on one of our iotedge but we are 2.8.3.1.
I've restarted all modules, removing the docker pods to see clearer.
At the first publish, I have this error:
```
[2022-08-12 08:15:34.390 ERR OpcUa] Publish #1, Reconnecting=False, Error: BadTimeout
[2022-08-12 08:15:34.406 ERR OpcUa] PUBLISH #1 - Unhandled error BadTimeout during Publish.
Opc.Ua.ServiceResultException: BadTimeout
   at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
   at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
```
Can you help ?

cristipogacean, 2022-08-12:
Hello @aleducq,

Can you share some more details on the configuration in this case? so we can understand better the context. For the same reason, it would be helpful if you could share a bigger part of the logs.

I'm not sure it is the same issue as above since this happens on publish #1, that is supposed to transport the initial data-change. 

Thank you!

aleducq, 2022-08-12:
I've sent you an email with logs and more details...