Azure/Industrial-IoT #1503: Use existing iot-Hub for iiot deployment

Padu98, 2021-12-19:
I recently deployed the Industrial-IoT Platform and its components with the deployment script. I now want to use a diffrent IoT-hub (free tier) and delete the autogenerated one. How can i connect the opc-cloud microservices and the engineering tool after those steps to my new IoT-hub. And if that is not possible, is there any way to use an existing IoT-Hub in the deployment script?

cristipogacean, 2021-12-21:
The parameters for of the various Azure resources in the the iiot deployment are stored in the deployment's key-vault. If you want to use a different IoT Hub, you need to change a couple of secrets in this key-vault: pcs-iothub-connstring and pcs-iothub-eventhubendpoint to point to the new IoT Hub connection string respective endpoint. Also make sure the new IoTHub has the following consumer groups defined: 
[image]

Unfortunately there is not much documentation, but I hope the info above will help.

cristipogacean, 2022-04-01:
Closing for now. Please reopen if needed.

closed 2022-04-01 by cristipogacean