Azure/Industrial-IoT #1504: Direct Methods Reference

Padu98, 2021-12-19:
Is there any documentation about the direct methods which can be used to access the edge modules? I just deployed the minimal srcript and was wondering how to call the twin moduls without the opc-cloud-microservices running. I know that there is a lot of documentation about the rest api which cant be to far off the direct mehtod calls.

mike-romanowski, 2021-12-20:
I have the same question...  We would like to use as many of the "off the shelf" modules supplied by Microsoft as possible however we have a requirement to be able to operate for up to 10 days without an internet connection.  For this reason we cannot rely on running our business logic against the OPC Twin Micorservice REST API, which to my knowledge, can only be deployed in the Azure cloud.  We are currently using the OPC Publisher module for telemetry but now need to implement the control side of our application.  Is it possible to access the OPC Twin module locally from the edge hub/device it is running on?  If so is there documentation that describes how this can be accomplished?

hansgschossmann, 2021-12-21:
@Padu98 , @mike-romanowski we have no plans to document the direct method interface of the modules which are part of the Industrial-IoT platform. those are part of an E2E solution and built to work in this setup.

We see a need for a direct method interface of the industrial modules and will address this in the near future:
- OPC Publisher will support a DM interface in standalone mode in the next version of the Industrial-IoT Platform (but not in orchestrated mode, which is the E2E use case i mentioned above)
- In the near future there will be a module available, which supports OPC Twin like functionality in standalone mode.

Can you both let me know which specific functionality you are looking for?

hansgschossmann, 2022-01-11:
@mike-romanowski  thanks for your answers. we plan to have something at private preview quality in the next 6 months for OPC Twin like functionality. in the same timeframe there will also be the update to OPC Publisher available.

If i do not reply to this issue till end of march, pls ping me here for an update.

emulic, 2022-12-04:
@hansgschossmann any news or maybe ETA for the standalone OPC Twin like module?

hansgschossmann, 2022-12-05:
@emulic we have changed our roadmap meanwhile and there is no OPC Twin like standalone module planned as part of the Industrial-IoT platform.

marcschier, 2023-06-22:
The directmethods API of OPC Publisher 2.9.0 preview 4 is documented in the docs/ folder: https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/readme.md#programming-against-opc-publisher-using-the-opc-publisher-api. Please see https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/features.md for support and feature state of each API and transport.

closed 2023-06-22 by marcschier