Azure/Industrial-IoT #1506: Discovery module: "Transient network error", "name does not resolve"

mwiot, 2021-12-20:
Hi, we have ran into an issue that we have not been able to figure out. A registered edge device in the iot hub does not show up in the engineering tool. When checking the discovery module logs i see that it reports an error: **Transient network error/ name does not resolve**. Any idea what is causing this? No errors are reported in the other modules. 

Using 
IIOT 2.8.1 (for all in one services and all modules)
Ubuntu 20.04

Hope you can give some insight in this issue. 
Best regards

```
[14:07:37 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Cancelling all pending requests...
[14:07:37 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Pending requests cancelled...
[14:07:37 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Stopped discovery processor.
[14:07:37 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to simulator-loki using MqttOverTcp.
[14:07:37 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Starting discovery processor...
[14:07:37 INF Root] Starting module OpcDiscovery version 2.8.1.1.
[14:07:37 INF Root] Initiating prometheus at port 9700/metrics
[14:07:37 INF Root] Prometheus metric server started.
[14:07:37 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[14:07:37 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context.
[14:07:38 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module simulator-loki_discovery disconnected due to Communication_Error - now Disconnected_Retrying...
[14:12:44 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module simulator-loki_discovery disconnected due to Retry_Expired - now Disconnected...
[14:12:44 ERR Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host failed to start.
[14:12:44 ERR Root] Error during module execution - restarting!
System.AggregateException: One or more errors occurred. (Transient network error occurred, please retry.)
 ---> Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: Transient network error occurred, please retry.
 ---> System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (00000005, 0xFFFDFFFF): Name does not resolve
   at System.Net.Dns.InternalGetHostByName(String hostName)
   at System.Net.Dns.ResolveCallback(Object context)
--- End of stack trace from previous location where exception was thrown ---
   at System.Net.Dns.HostResolutionEndHelper(IAsyncResult asyncResult)
   at System.Net.Dns.EndGetHostAddresses(IAsyncResult asyncResult)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenInternalAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass27_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass38_0.<<OpenInternalAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
   at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.ModuleClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, String moduleId, ITransportSettings transportSetting, TimeSpan timeout, IRetryPolicy retry, Action onConnectionLost, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Default\IoTSdkFactory.cs:line 280
   at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func`1[] options)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func`1[] options) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Core\src\Utils\Try.cs:line 106
   at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.CreateAsync(String product, IProcessControl ctrl) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Default\IoTSdkFactory.cs:line 193
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost.StartAsync(String type, String siteId, String productInfo, String version, IProcessControl reset) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Framework\src\Hosting\ModuleHost.cs:line 118
   at Microsoft.Azure.IIoT.Modules.Discovery.ModuleProcess.RunAsync() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.Discovery\src\ModuleProcess.cs:line 103
[14:12:44 INF Root] Stopped prometheus metric server
[14:12:44 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Cancelling all pending requests...
[14:12:44 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Pending requests cancelled...
[14:12:44 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Stopped discovery processor.
[14:12:44 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to simulator-loki using MqttOverTcp.
[14:12:44 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Starting discovery processor...
[14:12:44 INF Root] Starting module OpcDiscovery version 2.8.1.1.
[14:12:44 INF Root] Initiating prometheus at port 9700/metrics
[14:12:44 INF Root] Prometheus metric server started.
[14:12:44 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[14:12:44 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context.
[14:12:45 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module simulator-loki_discovery disconnected due to Communication_Error - now Disconnected_Retrying...
[14:17:51 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module simulator-loki_discovery disconnected due to Retry_Expired - now Disconnected...
[14:17:51 ERR Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host failed to start.
[14:17:51 ERR Root] Error during module execution - restarting!
System.AggregateException: One or more errors occurred. (Transient network error occurred, please retry.)
 ---> Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: Transient network error occurred, please retry.
 ---> System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (00000005, 0xFFFDFFFF): Name does not resolve
   at System.Net.Dns.InternalGetHostByName(String hostName)
   at System.Net.Dns.ResolveCallback(Object context)
--- End of stack trace from previous location where exception was thrown ---
   at System.Net.Dns.HostResolutionEndHelper(IAsyncResult asyncResult)
   at System.Net.Dns.EndGetHostAddresses(IAsyncResult asyncResult)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenInternalAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass27_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   --- End of inner exception stack trace ---

```

cristipogacean, 2021-12-21:
It seems to me that the network setting in the container create options of the discovery module prevent (for some reason that I don't know) name resolution for the iothub connection string. By default the discovery module is set in network host mode. An Idea is to start discovery in bridge network mode, like the publisher and twin modules are started - basically remove the network related options from the container start options and run with default network settings.

mwiot, 2021-12-22:
Hi thanks for the reply. Working now. 
Changed from

```
{ 
  "Hostname": "discovery", 
  "Cmd": [ 
    "PkiRootPath=/mount/pki" 
  ], 
  "NetworkingConfig": { 
    "EndpointsConfig": { 
      "host": {} 
    } 
  }, 
  "HostConfig": { 
    "NetworkMode": "host", 
    "CapAdd": [ 
      "NET_ADMIN" 
    ], 
    "Binds": [ 
      "/mount:/mount" 
    ] 
  } 
} 
```

to 

```
{ 
  "Hostname": "discovery", 
  "Cmd": [ 
    "PkiRootPath=/mount/pki" 
  ], 
  "HostConfig": { 
    "Binds": [ 
      "/mount:/mount" 
    ] 
  } 
} 
```

cristipogacean, 2021-12-22:
thanks for the feedback!

closed 2021-12-22 by cristipogacean

DidierDrogba2, 2022-02-15:
@cristipogacean What exactly is the purpose of running the Discovery Container in the host network anyway? Is it relevant for performing the Scans?