Azure/Industrial-IoT #1511: add modules to the edge device

DidierDrogba2, 2021-12-22:
Hi, I have some issues with my edge Device. I deployed the IIoT Platform using the full deployment-script. Additionally i created an edge device for the genarated iot-hub.  I did that by just copying the following in the device twin: 

    "tags": {
        "__type__": "iiotedge",
        "os": "Linux"
    }

After that I got the following Modules auto-generated:
twin
discovery
publisher
metricscollector

The next goal for me was to parse incoming data on my edge device. Therefore I created an Edge-analytics-job to run some tasks on the edge. The Idea for this step came from the "overview architecture" in the iiot-github-repo, in which the Analytic-Job is marked as optional.
Unfortunally I had to realize that every time I deployed the job on the edge, the whole device got overwritten by the default configuration.
My question is now if there is a way that my edge device is not getting updated every time I make some changes to it? Or is it possible to change  the default configuration of the device?
I also wanted to make some changes to the publisher following this tutorial: https://docs.microsoft.com/de-de/azure/industrial-iot/tutorial-data-explorer-import  which got overwritten afterwords. Or is this only in standalone mode possible?

cristipogacean, 2021-12-22:
Hi @DidierDrogba2, 
In order to disable the layered deployment you need to mark this device as "unmanaged" by IIoT Platform. Then the changes you make on the modules will be kept. Nevertheless, as long as the edge device contains the publisher/twin/discovery modules as well as the appropriate routing, the device will be recognized by the IIoT Platform and can be operated normally. 

```
...
"tags": {
    "__type__": "iiotedge",
    "os": "Linux",
    "unmanaged": true
}
...
```

If you want to enroll back the iotedge device into the IIoTPlatform and enable layered deployment, you need to remove the  "unmanaged" property by setting it to null e.g. "unmanaged": null

More details on edge deployment [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-install-iot-edge.md).

DidierDrogba2, 2021-12-22:
@cristipogacean thanks for the quick reply!

hansgschossmann, 2022-01-11:
@DidierDrogba2 looks like this issue is resolved. Please reactivate if you still see issues.

closed 2022-01-11 by hansgschossmann