Azure/Industrial-IoT #1521: OPC Twin Microservice

Padu98, 2022-01-13:
Hi there, I refer to the information about the OPC Twin Microservice on https://github.com/Azure/Industrial-IoT/blob/main/docs/services/twin.md .

When I write a "desired property" from the OPC twin microservice towards an Endpoint Identity - will it create a corresponding desired property in the device twin for the Endpoint Identity in the IoT Hub? To me it would seem logical to do that in order to support scenarios where the OPC UA Server or the IoT edge device are offline.

marcschier, 2022-01-14:
Hi @Padu98, you are correct, this would indeed be the way to go, but the Azure IoT Device twin is effectively a JSON document in a no-SQL database and is very small (properties together as JSON must be less than I believe 64k).  For this scenario it would quickly reach its limits.  Right now all transactions are through direct methods instead, which unfortunately require the edge to be online.  The right approach for most command control scenarios is RPC based on durable messaging, and we are working to get there mid to long-term, but we do not have a date for when this will be.

Padu98, 2022-01-16:
Hi @marcschier, 
Thanks for the quick reply!
In the Documentation it is also mentioned that WRITING a value on a Variable node is exposed as Desired/Reported Property on the “endpoint identity” on the Edge-Device. Does that mean that the “desired properties” of the Endpoint Identity running on the Edge are set by direct methods?
If I get this right it should at least be possible to communicate when the OPC-Server, connected to the Edge-Device, is currently offline? But in the REST API Reference under every WRITE-Request it is mentioned that “the endpoint must be activated and connected”. So how can I set the desired properties (Node values) of an endpoint identity that is “activated and connected” but currently “not reachable”? 

I am a little bit confused here. A Device Twin Pattern whether it is on the Edge or Cloud would not make any sense to me if it cant be used?

I refer to the information given in those two articles: 
https://github.com/Azure/Industrial-IoT/blob/main/docs/services/twin.md
https://github.com/Azure/Industrial-IoT/blob/main/docs/api/twin/readme.md

marcschier, 2022-01-18:
Hi @Padu98, I am truly sorry, but the documentation in twin.md is incorrect and from it the confusion stems.  I have just submitted an update to the documentation in PR #1525 to remove the reference to desired/reported properties.  Reading and writing of OPC UA values (and attributes) is supported via the REST API referred to in the second link.  You are correct that the REST API just translates into direct method calls on the twin (not desired properties though).  Therefore the twin must be online for the API call to work.  As mentioned, I think your scenario is absolutely valid, but unfortunately it is not yet supported.

marcschier, 2023-01-16:
We are not planning to support this feature.

closed 2023-01-16 by marcschier