Azure/Industrial-IoT #1528: deployment "killed" and bash does not response

linexc, 2022-01-23:
**Describe the bug**
I am trying building a connection between OPC UA and Azure and thereofore use the tutorial from [Ingesting OPC UA data with Azure Digital Twins](https://docs.microsoft.com/en-us/azure/digital-twins/how-to-ingest-opcua-data?tabs=portal). When I followed the tutorial step by step until starting the deployment. Since it is on Ubuntu 18, I gave the command **./deploy.sh**. But after I specified the name for my application, the bash crashed automatically after several seconds. 
Of course I have already two virtual systems. 

[image]
[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. cd Industrial-IoT
2. Input my actual existing resource group name
3. Inpout test as an application name 
4. Hit enter.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

hansgschossmann, 2022-01-25:
@linexc using deploy.sh from this repository is most likely installing much more than you are expecting. this will install the full Industrial-IoT platform which includes a bunch of microservices and is not what is described [here](https://docs.microsoft.com/en-us/azure/digital-twins/how-to-ingest-opcua-data?tabs=portal).

The best description of how to deploy OPC Publisher in standalone mode can be found [here](https://docs.microsoft.com/en-us/azure/industrial-iot/tutorial-publisher-deploy-opc-publisher-standalone).

Please let us know if this solves your problem.

hansgschossmann, 2022-02-01:
@linexc  were my comments above helpful in any way?

hansgschossmann, 2022-02-08:
@linexc pls reactivate if you still have questions.

closed 2022-02-08 by hansgschossmann