Azure/Industrial-IoT #1547: opcpublisher disconnected due to Communication_Error - now Disconnected_Retrying...

jaypaddy, 2022-02-04:
OPCPublisher is unable to connect to OPC-UA Server and is raising an error:

[19:47:41 INF Root] Starting module OpcPublisher version 2.8.1.2. 
[19:47:41 INF Root] Initiating prometheus at port 9702/metrics 
[19:47:41 INF Root] Prometheus metric server started. 
opcstacktracemask set to: 0x0
[19:47:41 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[19:47:41 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context. 
[19:47:46 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module <DEVICEID>_opcpublisher disconnected due to Communication_Error - now Disconnected_Retrying... 

We checked connectivity between IoT Edge and OPC UA Server and it was successful. We are using ip address to connect for OPC UA Server. However OPC Publisher seems to have a challenge and we are unable to resolve.

**iotedge check**

Configuration checks (aziot-identity-service)
---------------------------------------------
√ keyd configuration is well-formed - OK
√ certd configuration is well-formed - OK
√ tpmd configuration is well-formed - OK
√ identityd configuration is well-formed - OK
√ daemon configurations up-to-date with config.toml - OK
√ identityd config toml file specifies a valid hostname - OK
‼ aziot-identity-service package is up-to-date - Warning
    Installed aziot-identity-service package has version 1.2.4 but 1.2.5 is the latest stable version available.
    Please see https://aka.ms/aziot-update-runtime for update instructions.
√ host time is close to reference time - OK
√ preloaded certificates are valid - OK
√ keyd is running - OK
√ certd is running - OK
√ identityd is running - OK
√ read all preloaded certificates from the Certificates Service - OK
√ read all preloaded key pairs from the Keys Service - OK
√ ensure all preloaded certificates match preloaded private keys with the same ID - OK

Connectivity checks (aziot-identity-service)
--------------------------------------------
‼ host can connect to and perform TLS handshake with iothub AMQP port - Warning
    Could not retrieve iothub_hostname from provisioning file.
    Please specify the backing IoT Hub name using --iothub-hostname switch if you have that information.
    Since no hostname is provided, all hub connectivity tests will be skipped.
‼ host can connect to and perform TLS handshake with iothub HTTPS / WebSockets port - Warning
    Could not retrieve iothub_hostname from provisioning file.
    Please specify the backing IoT Hub name using --iothub-hostname switch if you have that information.
    Since no hostname is provided, all hub connectivity tests will be skipped.
‼ host can connect to and perform TLS handshake with iothub MQTT port - Warning
    Could not retrieve iothub_hostname from provisioning file.
    Please specify the backing IoT Hub name using --iothub-hostname switch if you have that information.
    Since no hostname is provided, all hub connectivity tests will be skipped.
√ host can connect to and perform TLS handshake with DPS endpoint - OK

Configuration checks
--------------------
√ aziot-edged configuration is well-formed - OK
√ configuration up-to-date with config.toml - OK
√ container engine is installed and functional - OK
√ configuration has correct URIs for daemon mgmt endpoint - OK
‼ aziot-edge package is up-to-date - Warning
    Installed IoT Edge daemon has version 1.2.5 but 1.2.7 is the latest stable version available.
    Please see https://aka.ms/iotedge-update-runtime for update instructions.
√ container time is close to host time - OK
‼ DNS server - Warning
    Container engine is not configured with DNS server setting, which may impact connectivity to IoT Hub.
    Please see https://aka.ms/iotedge-prod-checklist-dns for best practices.
    You can ignore this warning if you are setting DNS server per module in the Edge deployment.
        caused by: Could not open container engine config file /etc/docker/daemon.json
        caused by: No such file or directory (os error 2)
‼ production readiness: container engine - Warning
    Device is not using a production-supported container engine (moby-engine).
    Please see https://aka.ms/iotedge-prod-checklist-moby for details.
‼ production readiness: logs policy - Warning
    Container engine is not configured to rotate module logs which may cause it run out of disk space.
    Please see https://aka.ms/iotedge-prod-checklist-logs for best practices.
    You can ignore this warning if you are setting log policy per module in the Edge deployment.
        caused by: Could not open container engine config file /etc/docker/daemon.json
        caused by: No such file or directory (os error 2)
‼ production readiness: Edge Agent's storage directory is persisted on the host filesystem - Warning
    The edgeAgent module is not configured to persist its /tmp/edgeAgent directory on the host filesystem.
    Data might be lost if the module is deleted or updated.
    Please see https://aka.ms/iotedge-storage-host for best practices.
× production readiness: Edge Hub's storage directory is persisted on the host filesystem - Error
    Could not check current state of edgeHub container
        caused by: docker returned exit code: 1, stderr = Error: No such object: edgeHub
‼ Agent image is valid and can be pulled from upstream - Warning
    skipping because of previous failures

Connectivity checks
-------------------
‼ container on the default network can connect to upstream  AMQP port - Warning
    skipping because of previous failures
‼ container on the default network can connect to upstream HTTPS / WebSockets port - Warning
    skipping because of previous failures
‼ container on the default network can connect to upstream MQTT port - Warning
    skipping because of previous failures
‼ container on the IoT Edge module network can connect to upstream AMQP port - Warning
    skipping because of previous failures
‼ container on the IoT Edge module network can connect to upstream HTTPS / WebSockets port - Warning
    skipping because of previous failures
‼ container on the IoT Edge module network can connect to upstream MQTT port - Warning
    skipping because of previous failures
20 check(s) succeeded.
9 check(s) raised warnings.
1 check(s) raised errors.
7 check(s) were skipped due to errors from other checks.

cristipogacean, 2022-02-04:
This publisher logged error:
> [19:47:46 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module _opcpublisher disconnected due to Communication_Error - now Disconnected_Retrying...

refers to the upstream connectivity to the IoT hub. It has nothing to do with the connectivity to the OPC UA Server.

To me it looks like the IoT Edge runtime is not running properly. If troubleshooting/reinstalling the runtime does not help, I would recommend you try with IoT edge 1.1 (LTS) as well.

jaypaddy, 2022-02-04:
Thank you @cristipogacean. You are correct. it was a problem with edgeAgent and as we debug we found the issue and got it resolved.... OPCPublisher is working fine now.

closed 2022-02-04 by jaypaddy