Azure/Industrial-IoT #1548: Support OPC UA Historical Access implemented?

mpaxh, 2022-02-04:
Does OPC Publisher support part 11 (Historical Access) of the opc ua specification? The documentation does not detail which release / parts of the opc ua specification are implemented. If yes, an example would be great.

cristipogacean, 2022-02-04:
OPC UA Historical Access is out of the scope of OPC Publisher. HA is integrated in the OPC Twin module and can only be used together with the IIoT platform microservices.

Haeavar, 2022-02-07:
Hi @cristipogacean 
Since the IIoT platform microservices will be discontinued. Are there any plans to support HA in the future? Do you have any detailed information about HA and the IIoT platform microservices?

hansgschossmann, 2022-02-08:
@Haeavar i am interested to understand your use case for HA in more detail. Can you share those here?

mpaxh, 2022-02-08:
@hansgschossmann If the connection to the opc ua server is interrupted for whatever reason (e.g. network not available) a gap in the collected data by OPCPublisher results. However, typically the system on which the opc ua server runs stores the data for some period of time. HA could be a way to backfill the gaps in the data once the connection to the OPC UA server is restored.

hansgschossmann, 2022-02-15:
@mpaxh thank you for your feedback. we are also discussing to support enabling queue size on MonitoredItems. Would this be enough for you? HA would definitely provide data for longer dropouts if the server supports it.

mpaxh, 2022-02-15:
Not sure, what would be the max queue size be? We would like to be able to backfill at least 5 days, for around 20k variables per server with the fastest variables sampling at 1 Hz. More than 5 days are welcome, that would make fixing the connection problem a bit more relaxed ;)
For reference, our servers provide the full history for 13 weeks.

hansgschossmann, 2022-02-22:
@mpaxh Queuesize is most likely too small for 5 days. As @cristipogacean pointed out in his first answer this is a feature which would require to use the full Industial-IoT platform (including microservices) and using the OPC Twin REST API to access HA data.

mpaxh, 2022-02-22:
@hansgschossmann @cristipogacean ok, since the iiot platform microservices will be discontinued, what is the solution for HA then?

hansgschossmann, 2022-03-01:
@mpaxh i am tagging this as feature request for OPC Publisher and will follow-up internally with the team.

Haeavar, 2022-03-04:
> @Haeavar i am interested to understand your use case for HA in more detail. Can you share those here?

@hansgschossmann To safe costs and bandwidth, we want to send historical opcua data via file transfer to the cloud and only a few most important nodes via OpcPublisher live stream.

hansgschossmann, 2022-03-08:
@Haeavar then you would pull the HA data from the server in your custom module which is doing the file upload when you are missing some datapoints?

marcschier, 2023-05-19:
OPC Publisher 2.9.0 preview 2 has HA API in preview, documentation will be provided later but using the openapi.json file in the opc-publisher docs folder it one could piece it together.  This is preview mode and only supported though GitHub issues.

closed 2023-05-19 by marcschier