Azure/Industrial-IoT #1572: Expose the local OPC UA servers transparently in the platform

saramonteiro, 2022-02-14:
**Describe the solution you'd like**
Some SCADA systems, like [System Platform](https://www.aveva.com/en/products/system-platform/), has a built-in OPC-UA client that connects directly to OPC-UA servers to monitor and control assets.
When these SCADA systems are on premise in the same equipments' network, it's quite simple to connect them using the endpoints URLs. But if the SCADA system is on a remote private or public cloud, it may require a VPN connection, which may be an expensive approach.
The current state of the IIoT Platform doesn't provide a direct way to connect to this remote SCADA system given in the example. To make it possible, I understand we'd need to develop an OPC-UA interface that exposes the OPC-UA servers from the shop floor on a transparent way, making use of the APIs exposed by the microservices. Please, take a look in the following image that I prepared to illustrate the approach:

[image]

So, the discussion (feature request) I'd like to bring here is to create this "interface module" and make it available also together with the IIoT Platform. Or develop another approach to achieve the same goal.

Thanks in advance to open this space and for the Open Source Project.

hansgschossmann, 2022-02-15:
@saramonteiro we had in the past done a similar implementation of this, but removed it because of limited customer request and involved maintenance cost. Way back we had done what was called OPC Proxy. This was made of a client part and a modified OPC Stack. If you look on github you may still find all of those implementations.

saramonteiro, 2022-02-15:
> @saramonteiro we had in the past done a similar implementation of this, but removed it because of limited customer request and involved maintenance cost. Way back we had done what was called OPC [...]

Hi @hansgschossmann can you be more specific about where I can find these stuff?
Maybe, which release this was still present?

hansgschossmann, 2022-02-22:
here is a link to the old OPC Proxy module. The gateway service in Industrial-IoT was experimental in 2.7.0:: https://github.com/Azure/Industrial-IoT/blob/release/2.7/docs/services/twin-gateway.md
It is still available as a container in 2.8.1, but no longer supported or recommended to use.

closed 2022-02-22 by hansgschossmann