Azure/Industrial-IoT #1578: Discovery docker network

DidierDrogba2, 2022-02-16:
Hi,
I have a question regarding the Discovery Modul on the Gateway. What exactly is the reason for the Container running in host-network? All the other Container don't do so. Is it necessary to perform the Scans?

cristipogacean, 2022-02-16:
Hi.

The reason for this is ease-of-use only.

When the scan addresses range is not specified in the discover request, the discovery module looks for all the network adapters and performs scans in the address ranges associated to the respective network adapters. This way, discovery module will only be able to access the edge appliance's peer opc ua servers when the module uses the host network. 

When the scan addresses range is specified in the discover request, or when opc ua discovery urls are used, the discovery module can very well reside on the internal container default network, like the other modules do. This will not affect the scan at all.

hansgschossmann, 2022-02-22:
@cristipogacean thanks for providing the information. I will close this issue. @DidierDrogba2  pls reactivate if you need further information.

closed 2022-02-22 by hansgschossmann