Azure/Industrial-IoT #1592: OPC publisher stop sending after subscribing to new tags

aleducq, 2022-02-23:
**Describe the bug**
OPC publisher was sending telemetry data normally then after adding a new tag, without any error messages, the telemetry stops.
We have several edges and it happens only this time on this edge. The differences between this edge and the others is this edge runtime is a 1.2 running on Ubuntu and others are in 1.1 running in Cent OS

**Screenshots**
an extract from the publisher logs before and after the event (I can send you the support-bundle extraction but I don't want it to be public)

`[12:27:22 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://XXXXXXXXXXXXXXX.com:49320
  # Ingestion duration                 :    08:20:35:00 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :        762,437 (1/s)
  # Ingress ValueChanges (from OPC)    :        764,535 (1/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :        764,535
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :        759,827
  # Encoder avg Notifications/Message  :              1
  # Encoder avg IoT Message body size  :            628 (0%)
  # Encoder avg IoT Chunk (4 KB) usage :            0.2
  # Estimated IoT Chunks (4 KB) per day:         85,782
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :        759,827 (0.99/s)
  # Connection retries                 :              0
 
[12:27:27 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Closing subscription uata6f41c4e7f569ae79f79deb8518a5a559355923e_1000 
[12:27:27 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'uata6f41c4e7f569ae79f79deb8518a5a559355923e_1000' unregistered from session '389d1bdc7ce9fffaa82bda24946576c9fb5b197a' in state Running 
[12:27:28 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker+JobProcess] Job uata6f41c4e7f569ae79f79deb8518a5a559355923e cancelled. 
[12:27:28 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 183: Device DataSetWriterV2_uata6f41c4e7f569ae79f79deb8518a5a559355923e disconnected due to Client_Close - now Disabled... 
[12:27:28 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Closing subscription uata6f41c4e7f569ae79f79deb8518a5a559355923e_1000 
[12:27:28 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Processing job continuation... 
[12:27:28 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to sl79v535 using MqttOverTcp. 
[12:27:28 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[12:27:28 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Device DataSetWriterV2_uata6f41c4e7f569ae79f79deb8518a5a559355923e reconnected due to Connection_Ok. 
[12:27:28 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'uata6f41c4e7f569ae79f79deb8518a5a559355923e_1000' registered/updated in session '389d1bdc7ce9fffaa82bda24946576c9fb5b197a' in state Running 
[12:27:28 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 3 monitored items to subscription uata6f41c4e7f569ae79f79deb8518a5a559355923e_1000 
[12:27:28 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Updated 3 monitored items in subscription uata6f41c4e7f569ae79f79deb8518a5a559355923e_1000 
[12:27:28 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'uata6f41c4e7f569ae79f79deb8518a5a559355923e_1000' registered/updated in session '389d1bdc7ce9fffaa82bda24946576c9fb5b197a' in state Running 
[12:28:28 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://XXXXXXXXXXXXXXX.com:49320
  # Ingestion duration                 :    00:00:00:59 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :              1 (0.02/s)
  # Ingress ValueChanges (from OPC)    :              3 (0.05/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :              3
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :              1
  # Encoder avg Notifications/Message  :              3
  # Encoder avg IoT Message body size  :          1,857 (1%)
  # Encoder avg IoT Chunk (4 KB) usage :            0.5
  # Estimated IoT Chunks (4 KB) per day:          1,464
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :              1 (0.02/s)
  # Connection retries                 :              0`

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.3 LTS (GNU/Linux 5.4.0-91-generic x86_64)
 - Version 2.7.206
 - Thingworx connector 4.2
 - Thingworx 9.0.8

cristipogacean, 2022-02-24:
From the logs here I can only see that after the reconfiguration the server is only sending the initial datachange for the 3 monitored items within the timeframe of 59 seconds. Could it be that the 3 monitored items are "constant" in this timeframe, or the sampling interval for the 3 monitored items is set to something > 59 seconds?

BTW, Could you please try to update to the LTS version (2.8.x)? 2.7.x is going out of support.

aleducq, 2022-02-24:
Hello Cristian, thanks for your reply. No, one of the item is a simulated heartbeat tag from OPC server increasing every second to check the health of the communication. What you see above is a part of the log, but if I look all the logs, I have nothing sent until the restart of the publisher module after 2h45 of run:
```
[15:13:28 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://SW08P335.eu.seb.com:49320
  \# Ingestion duration                 :    00:02:45:59 (dd:hh:mm:ss)
  \# Ingress DataChanges (from OPC)     :              1 (0/s)
  \# Ingress ValueChanges (from OPC)    :              3 (0/s)
  \# Ingress BatchBlock buffer size     :              0
  \# Encoding Block input/output size   :              0 | 0
  \# Encoder Notifications processed    :              3
  \# Encoder Notifications dropped      :              0
  \# Encoder IoT Messages processed     :              1
  \# Encoder avg Notifications/Message  :              3
  \# Encoder avg IoT Message body size  :          1,857 (1%)
  \# Encoder avg IoT Chunk (4 KB) usage :            0.5
  \# Estimated IoT Chunks (4 KB) per day:              9
  \# Outgress Batch Block buffer size   :              0
  \# Outgress input buffer count        :              0
  \# Outgress input buffer dropped      :              0
  \# Outgress IoT message count         :              1 (0/s)
  \# Connection retries                 :              0
```
Regarding the version, we can't upgrade to the 2.8 version because we are using Thingworx OPC UA encapsulation and the last validated version by Thingworx editor PTC is the 2.7.206... which should be a LTS version but obviously it isn't.

cristipogacean, 2022-02-25:
@aleducq, thanks for validating this for us. I still think we don't have enough details to figure out what could cause this issue.
Would you be able to provide us the support bundle of this edge device? 
Would you be so kind and write down the steps to replicate this issue so we can try it out internally?

hansgschossmann, 2022-03-01:
@aleducq information on how to collect support-bundle information can be found at:  https://docs.microsoft.com/en-us/azure/iot-edge/troubleshoot?view=iotedge-2020-11

aleducq, 2022-03-01:
Hi @hansgschossmann, I've sent the support bundle information directly to @cristipogacean and @karok2m by email.

hansgschossmann, 2022-03-08:
@aleducq thanks for sending. @cristipogacean , @karok2m  have you had a chance to look at the support bundle?

karok2m, 2022-03-08:
@hansgschossmann, unfortunately OPC Publisher logs in the support bundle did not indicate any issues. I've asked @aleducq to enable debug logs for this deployment and send us new support bundle once the issue occurs again. It would also be useful to have the steps to replicate the problem. @aleducq, can you please add some information about that here ?

hansgschossmann, 2022-03-15:
@aleducq have you had a chance to replicate with debug logs enabled?

hansgschossmann, 2022-03-22:
I think that it is not possible to set it via command line for exactly one specific IoT Edge device if the config come from a  layered deployment. You would need to author a deployment manifest for this edge to add the --ll. But i think there is also an option to set it via the module twin at least for OPC Twin and Discovery modules. For OPC Publisher: @cristipogacean and @karok2m can you help here?

karok2m, 2022-03-25:
@aleducq, you can add `--ll=debug` to `Cmd` options of deployment manifest if you have a separate one for deployment of this particular OPC Publisher.

If OPC Publisher is running in orchestrated mode, which I believe is your case, then you can do this for a particular instance either via (1) adding `"LogLevel": "Debug"` to desired properties of module identity twin in IoT Hub or via (2) REST APIs which will do this for you. For REST APIs, you can do this using `PATCH` call on `/registry/v2/publishers/{publisherId}` endpoint from Swagger of OPC Registry microservice:

[image]

hansgschossmann, 2022-04-12:
no activity since several weeks: closing

@aleducq pls reactivate if more help is needed.

closed 2022-04-12 by hansgschossmann