Azure/Industrial-IoT #1610: Node publishing doesn't work for some OPC UA endpoints

emulic, 2022-03-02:
**Describe the bug**
We are running an Edge device with IIoT modules in orchestrated mode in the version 2.8. 
Discovery feature of the IIoT solution  found 9 OPC UA endpoints in the customer network and we activated node publishing for 5 endpoints without any issues.
Unfortunately, the publishing doesn't work for the 6th configured OPC UA endpoint.
During the configuration everything looked fine but we are not receiving any messages in the IoT Hub from this 6th OPC UA endpoint. 
There are no errors or warnings in the module logs.
We observed, that our Edge device in the IoT Hub is missing DataSetWriter device identity for this 6th OPC UA endpoint. However, we found the corresponding DataSetWriter device in the IoT Hub but somehow the parent-child relation to the Edge device is missing. 
We tried  to add the DataSetWriter device to the edge device manually by using Azure Portal, but this didn't work, since our edge device doesn't appear in the list of devices selectable as parent. We also tried in opposite direction, to add the DataSetWriter device as child to our Edge device but we see the same issue there, the DataSetWriter is not selectable as child device.

Multiple times we tried to turn off published nodes and deactivate the OPC endpoint and configured it again with no success. 
We also tried to delete the job and cleanup device registrations in IT Hub and to start the job again by using Publisher REST Api but this also didn't work. 

It looks like the solution has some kind of limitation to max 5 OPC UA endpoints. To check that, we removed one of the working 5 endpoints and then tried to add the endpoint which didn't work before and suddenly this endpoint properly published nodes.
After that we tried to add the removed endpoint again as 6th endpoint to the Edge device but we observed the same issue as described above.

**To Reproduce**
Steps to reproduce the behavior:
1. Add 5 OPC UA endpoints and configure publishing of at least one node per endpoint by using Engineering tool.
2. Add 6th OPS UA endpoint and configure publishing of at least one node by using Engineering tool.
3. Go to the "Published nodes" of the 6th OPC UA endpoint and check if the value of the published is changing. 
You'll observe, that the value is not changing. The published nodes values of other 5 OPC UA endpoints are properly published. This can also be checked by connection to the IoT Hub by using VS, VS Code , Iot Explorer or other tools.

**Expected behavior**
Publishing of node values should work for all configured OPC UA endpoints, not only for first 5 configured endpoints.

hansgschossmann, 2022-03-08:
@cristipogacean can this be limited by the MaxWorker settings of PublisherConfigApiModel and then can be changed by the REST API call "Update publisher configuration" (documented here: https://github.com/Azure/Industrial-IoT/tree/main/docs/api/registry)?

@emulic if you have time, you can give it a shot as well.

cristipogacean, 2022-03-08:
@emulic, the default MaxWorker setting is set to 5, so one publisher will handle up to 5 publishing jobs. You can increase this value to a more appropriate value in the engineering tool as in the following print-screen:

[image]

Alternatively, the registry service api can be called as @hansgschossmann  already mentioned
e.g.: 
[image]

emulic, 2022-03-10:
Thanks @hansgschossmann & @cristipogacean, this worked.

closed 2022-03-10 by emulic