Azure/Industrial-IoT #1625: Iothub stops receiving data from edge device

lordalek, 2022-03-15:
Iothub stops receiving data from edge device. 
The $edgeHub container retries every five minutes but same result every time.

Having scrolled through the $edgeHub logs, there is nothing that indicates that there is a way to manually trigger this behavior.
The publisher container has no logs available from the incident.

Firstly we restarted the publisher container but the connection was not restored.
Restarting the $edgeHub container restored the connection.

```
<6> 2022-03-07 11:45:33.337 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-07 11:47:08.576 +00:00 [INF] - Started task to cleanup processed and stale messages for endpoint iothub
<6> 2022-03-07 11:47:08.598 +00:00 [INF] - Cleaned up 13 messages from queue for endpoint iothub and 13 messages from message store.
<6> 2022-03-07 11:50:33.336 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-07 11:50:33.336 +00:00 [INF] - Error authenticating token for kkv-edge-gateway/publisher because the token is expired or could not be parsed
<6> 2022-03-07 11:50:33.336 +00:00 [INF] - Credentials for client kkv-edge-gateway/publisher are not valid.
<4> 2022-03-07 11:50:33.336 +00:00 [WRN] - Reauthenticating client kkv-edge-gateway/publisher failed, removing client connection
<6> 2022-03-07 11:50:33.336 +00:00 [INF] - Unable to re-authenticate kkv-edge-gateway/publisher, dropping client connection.
<6> 2022-03-07 11:50:33.337 +00:00 [INF] - Removing device connection for device kkv-edge-gateway/publisher with removeCloudConnection flag 'True'.
<4> 2022-03-07 11:50:33.337 +00:00 [WRN] - Closing connection for device: kkv-edge-gateway/publisher, Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnectionException: Connection closed for device kkv-edge-gateway/publisher., 
<6> 2022-03-07 11:50:33.337 +00:00 [INF] - Disposing MessagingServiceClient for device Id kkv-edge-gateway/publisher because of exception - Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnectionException: Connection closed for device kkv-edge-gateway/publisher.
<6> 2022-03-07 11:50:33.337 +00:00 [INF] - Closing device proxy for device Id kkv-edge-gateway/publisher
<6> 2022-03-07 11:50:33.338 +00:00 [INF] - Closing receiver in cloud proxy 6ac16ecc-cdb9-4950-b209-f4e2867bef11 for kkv-edge-gateway/publisher
<6> 2022-03-07 11:50:33.338 +00:00 [INF] - Closed cloud proxy 6ac16ecc-cdb9-4950-b209-f4e2867bef11 for kkv-edge-gateway/publisher
<6> 2022-03-07 11:50:33.338 +00:00 [INF] - Device connection removed for device kkv-edge-gateway/publisher
<6> 2022-03-07 11:50:33.348 +00:00 [INF] - Error authenticating token for DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049 because the token is expired or could not be parsed
<6> 2022-03-07 11:50:33.349 +00:00 [INF] - Credentials for client DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049 are not valid.
<4> 2022-03-07 11:50:33.349 +00:00 [WRN] - Reauthenticating client DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049 failed, removing client connection
<6> 2022-03-07 11:50:33.349 +00:00 [INF] - Unable to re-authenticate DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049, dropping client connection.
<6> 2022-03-07 11:50:33.349 +00:00 [INF] - Removing device connection for device DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049 with removeCloudConnection flag 'True'.
<4> 2022-03-07 11:50:33.349 +00:00 [WRN] - Closing connection for device: DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049, Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnectionException: Connection closed for device DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049., 
<6> 2022-03-07 11:50:33.349 +00:00 [INF] - Disposing MessagingServiceClient for device Id DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049 because of exception - Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnectionException: Connection closed for device DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049.
<6> 2022-03-07 11:50:33.349 +00:00 [INF] - Closing device proxy for device Id DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049
<6> 2022-03-07 11:50:33.350 +00:00 [INF] - Closing receiver in cloud proxy c3f6cc4b-8c8e-455d-ba29-3c092322bd17 for DataSetWriterV2_uat812748106293fcf387e1c4922cb0e7bbf1b8c049
<6> 2022-03-07 11:50:33.697 +00:00 [INF] - Updated reported properties for kkv-edge-gateway/$edgeHub
<6> 2022-03-07 11:50:59.784 +00:00 [INF] - Starting periodic operation Get EdgeHub config...
<6> 2022-03-07 11:51:00.054 +00:00 [INF] - Obtained edge hub config from module twin
<4> 2022-03-07 11:51:00.056 +00:00 [WRN] - Error getting edge hub configuration.
System.ObjectDisposedException: The CancellationTokenSource has been disposed.
   at System.Threading.CancellationTokenSource.ThrowObjectDisposedException()
   at Microsoft.Azure.Devices.Routing.Core.Router.ReplaceRoutes(ISet`1 newRoutes) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Routing.Core/Router.cs:line 165
   at Microsoft.Azure.Devices.Edge.Hub.Core.Config.ConfigUpdater.UpdateRoutes(IReadOnlyDictionary`2 routes, Boolean replaceExisting) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Edge.Hub.Core/config/ConfigUpdater.cs:line 135
   at Microsoft.Azure.Devices.Edge.Hub.Core.Config.ConfigUpdater.<UpdateConfig>b__11_0(EdgeHubConfig ehc) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Edge.Hub.Core/config/ConfigUpdater.cs:line 110
   at Microsoft.Azure.Devices.Edge.Hub.Core.Config.ConfigUpdater.UpdateConfig(Option`1 edgeHubConfig) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Edge.Hub.Core/config/ConfigUpdater.cs:line 112
   at Microsoft.Azure.Devices.Edge.Hub.Core.Config.ConfigUpdater.PullConfig(Func`2 configGetter) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Edge.Hub.Core/config/ConfigUpdater.cs:line 93
<6> 2022-03-07 11:51:00.060 +00:00 [INF] - Successfully completed periodic operation Get EdgeHub config
<6> 2022-03-07 11:51:00.296 +00:00 [INF] - Updated reported properties for kkv-edge-gateway/$edgeHub
<6> 2022-03-07 11:55:33.335 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-07 11:55:33.336 +00:00 [INF] - Error authenticating token for kkv-edge-gateway/twin because the token is expired or could not be parsed
<6> 2022-03-07 11:55:33.336 +00:00 [INF] - Credentials for client kkv-edge-gateway/twin are not valid.
<4> 2022-03-07 11:55:33.336 +00:00 [WRN] - Reauthenticating client kkv-edge-gateway/twin failed, removing client connection
<6> 2022-03-07 11:55:33.336 +00:00 [INF] - Unable to re-authenticate kkv-edge-gateway/twin, dropping client connection.
<6> 2022-03-07 11:55:33.336 +00:00 [INF] - Removing device connection for device kkv-edge-gateway/twin with removeCloudConnection flag 'True'.
<4> 2022-03-07 11:55:33.336 +00:00 [WRN] - Closing connection for device: kkv-edge-gateway/twin, Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnectionException: Connection closed for device kkv-edge-gateway/twin., 
<6> 2022-03-07 11:55:33.336 +00:00 [INF] - Disposing MessagingServiceClient for device Id kkv-edge-gateway/twin because of exception - Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnectionException: Connection closed for device kkv-edge-gateway/twin.
<6> 2022-03-07 11:55:33.337 +00:00 [INF] - Closing device proxy for device Id kkv-edge-gateway/twin
<6> 2022-03-07 11:55:33.337 +00:00 [INF] - Closing receiver in cloud proxy a8acd6db-3a6d-40ad-8ae7-3b502278ec1d for kkv-edge-gateway/twin
<6> 2022-03-07 11:55:33.337 +00:00 [INF] - Closed cloud proxy a8acd6db-3a6d-40ad-8ae7-3b502278ec1d for kkv-edge-gateway/twin
<6> 2022-03-07 11:55:33.338 +00:00 [INF] - Device connection removed for device kkv-edge-gateway/twin
<6> 2022-03-07 11:55:33.700 +00:00 [INF] - Updated reported properties for kkv-edge-gateway/$edgeHub
<6> 2022-03-07 12:00:33.331 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-07 12:05:33.338 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
```
**Container versions:**
 - mcr.microsoft.com/azureiotedge-hub:1.1
 - mcr.microsoft.com/azureiotedge-agent:1.1
 - mcr.microsoft.com/iotedge/opc-twin:2.8.1
 - mcr.microsoft.com/iotedge/opc-publisher:2.8.1
 - mcr.microsoft.com/iotedge/discovery:2.8.1

hansgschossmann, 2022-03-15:
@lordalek the following two lines indicate that this seems to be an issue related to azureiotedge-hub:
Firstly we restarted the publisher container but the connection was not restored.
Restarting the $edgeHub container restored the connection.

Do you know if the latest 1.1.x version was pulled to you edge device (you can find this out when restarting edgeHub and checking its logs).

The root cause seems to be that edgeHub seems not to be able to renew auth tokens, which might be caused by some communication issues with the cloud.

lordalek, 2022-03-15:
Is it this line you mean?

```<6> 2022-03-14 08:34:40.595 +00:00 [INF] - Version - 1.1.8.48502653 (54b31809c98ee83e36679ecf92c5582de37172c4)```

hansgschossmann, 2022-03-15:
yep. 1.1.11 is the latest available. but i have not seen any significant changed in edgeHub since 1.1.8. can you cleanup the images on your edge and start over. before you are doing this, capture a support-bundle using the iotedge command. 
if you hit this issue again, pls fetch a support-bundle as soon as you get aware of the issue.

hansgschossmann, 2022-03-29:
We close this issue for now. Pls reactivate if you hit it again and provide a support-bundle.

closed 2022-03-29 by hansgschossmann