Azure/Industrial-IoT #1640: Script is not compatible with your computer

rploeg, 2022-03-21:
I have installed Industrial - IoT before, but on a new tenant I want to install the environment with the standard script. I am using the parameter -subscriptioname to select the right subscription of my user. But when the registration of AAD begins, I get this error:

[image]

The user is owner of the subscription and is global administrator of AAD. Any idea what is going wrong?

hansgschossmann, 2022-03-22:
@karok2m do you have an idea what can be the reason for this issue? @rploeg which version of the deployment script have you used.

rploeg, 2022-03-23:
The version of two days ago --> main. I have tried it on another Windows 11 machine and there it was working. So it looks like a problem on one of my machines.

closed 2022-03-23 by rploeg