Azure/Industrial-IoT #1641: Unable to connect to an OPC endpoint using Base256 with OPC Publisher

dagefjestad, 2022-03-21:
Hi

I have some trouble to connect to an OPC endpoint which use: 
•	Security policy: Basic256 
•	Message Security Mode: Sign & Encrypt 

We have exchange certificates - and  added it to thrusted. 
I am using Latest version of OPC Publisher. 

Error message: 
2022-03-21 15:14:41.418 +00:00 [INF] Azure IIoT b37d9b2f8445a4382e96f1d18342b2a48c455343 took "00:00:00.2354173".
2022-03-21 15:14:41.429 +00:00 [WRN] Failed to create session 'b37d9b2f8445a4382e96f1d18342b2a48c455343' due to One or more errors occurred. (Endpoint does not support the user identity type provided.)

pn.json: 
      "EndpointUrl": "opc.tcp://someIPadr:62550/DataAccessServer",
      "UseSecurity": true,
      "OpcAuthenticationMode" : "Anonymous",

Is there any way to more specific for which security policy to use? Is there something else that could be wrong?

Thanks,
Dag Efjestad

dagefjestad, 2022-03-22:
Hi,
Yes - it sounds like this but I tested the endpoint with UAExpert (Client tool) with anonymous - that works.

hansgschossmann, 2022-03-29:
@dagefjestad just to be sure: you are using OPC Publisher 2.8.2?
cc: @cristipogacean

dagefjestad, 2022-03-29:
Thank you for the clarification about UserName. 

When I dont have something in the UserName the Server respond with "Endpoint does not support the user identity type provided" very early in the negotiation. I asked the server guy about the password - they dont use it. They say that UserName should not be set but they cant explain why I get the identity type error and when I have some value I get further in negotiation before it stops with "BadSessionNotActivated" ?! 

I mange to connect to the server with UAExpert - but the strange here is that I get an error: "BadCertificateHostnameInvalid" during connection fase - choose ignore to continue. 

Is there any connection between fields in the certificate and identity (user) or username. I have set Application Name equal to Common Name in the certificate - "Customer".  

The server is based on the Softing OPC/UA SDK.

hansgschossmann, 2022-03-30:
The "BadCertificateHostnameInvalid" message is good to ignore and not concerning. This has to do with the SubjectName in the certificate not matching the actual hostname in the Endpoint URL you use to connect.

Do you use any username/password when connecting with UA Expert or do you use no authentication (means anonymous)?

hansgschossmann, 2022-03-30:
Got it. With "UseSecurity": true, we choose the best Security policy/Security mode combination supported by OPC Publisher and the server. Basic256/Sign & encrypt might not be the most secure one, but there might be a more secure one. You should see all the combinations supported by the server in UA expert. Can you try to connect to the most secure one using anonymous?

dagefjestad, 2022-03-30:
I think the problem is that the server failing in negotiation. It is a "home made" server based on the Softing SDK. I have send a mail to the other party and ask them for a meeting with someone who knows the server software.

hansgschossmann, 2022-03-31:
Please let us know when zou have more details.

hansgschossmann, 2022-04-12:
closing because of no activity.

@dagefjestad pls reopen if there is more help needed.

closed 2022-04-12 by hansgschossmann