Azure/Industrial-IoT #1649: TSI no data displaying

rploeg, 2022-03-23:
Hi,

I am running the latest deployment script (main) and deployed the environment. I have setup a own Azure IoT Edge device (physical) and a OPC UA server. I successfully discovered the server and turned on the nodes:

[image]

The data is flowing into the Azure IoT Hub:

[image]

but dont' see any data flowing into TSI:

[image]

I have setup the data acces policy, but there is no data flowing in, so I don't see any data. Does the data showing in TSI only work with the example Edge devices?

hansgschossmann, 2022-03-24:
@rploeg as you can see the issue is because of the following error:
"Endpöoint deos not support the user identity type provided"
So it is because the OPC UA server does not support the user auth which is configured.

@dacolgit can you pls describe which user auth is supported by the Engineering tool?

hansgschossmann, 2022-03-25:
Do you see any logs on the OPC UA server side? They might give you an idea what is wrong? You can also check if changing the security policy/mode has some impact.

karok2m, 2022-04-04:
@rploeg, can you please restart the OPC Publisher and collect the logs from the beginning ? Having a bit more logs might give us a clearer picture.

hansgschossmann, 2022-06-21:
@rploeg  there was not response on this for quite a while. closing for now. please reactivate if the issue is still there.

closed 2022-06-21 by hansgschossmann