Azure/Industrial-IoT #1654: Azure IoT Edge OPC publisher Module Not sending the defined parameters

A-DavidJeyapaul, 2022-03-25:
Hi Team,

We are using IoT Edge OPC publisher Module to publish the data from our OPC server to IoT hub. Initially it was sending the data with the below schema.
[image]

And Now its not updated to as per the below one.
[image]

The difference between both of the schema is **ApplicationUri**  and **EndpointUrl**

We have multiple OPC servers and we are segregating the data based on OPC servers in Azure Stream Analytics Jobs. Since this was updated, our data segregation is gets affected and none of the data is getting processed by our Job.

_Thanks in advance for your support_.

cristipogacean, 2022-03-25:
@A-DavidJeyapaul, that's right, we fixed a regression in 2.8.2 related to the EdnpointUrl that leads to this behavior. If you need the ApplicationUri, you can get that adding a cli argument: --FullFeaturedMessage=true (--fm=true). This will add the application uri to the message as well. more details on the telemetry payload [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/dev-guides/telemetry-messages-format.md#standalone-publisher-mode---legacy).

A-DavidJeyapaul, 2022-03-29:
Thanks for your support @cristipogacean and apologize for delayed response.

closed 2022-03-29 by A-DavidJeyapaul