Azure/Industrial-IoT #1655: Breaking Change in OPC Publisher 2.8.2 in way how NodeId is reported back

derSchtefan, 2022-03-25:
**Describe the bug**
OPC Publisher 2.8.2 introduces an undocumented breaking change in which the NodeId is reported back in each measurement. When the subscription to a node is specified in the format `nsu=http://COMPANY.com/MS/Unit;i=6076`, versions prior to 2.8.2 reported back the id in the NodeId element as `http://COMPANY.com/MS/Unit#i=6076` (namespace, hash, id).

With 2.8.2 this was changed to report back `nsu=http://COMPANY.com/MS/Unit;i=6076` .

The issue here is that our existing pipelines are all configured to expect the format prior to 2.8.1, and now a whole bunch of mapping tables across devices and backend systems are unable to map the NodeId to its measurement.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a publishednodes.json specifying a NodeId like `nsu=http://COMPANY.com/MS/Unit;i=6076` 
2. Observe OPC Publisher 2.8.1 to return `http://COMPANY.com/MS/Unit#i=6076` (namespace, hash, id) as telemetry
3. Observe OPC Publisher 2.8.2 to return `nsu=http://COMPANY.com/MS/Unit;i=6076` (namespace, hash, id) as telemetry

**Expected behavior**
Even though the new behaviour seems "more correct", and the old behaviour or reporting a mangled name back seems to be "wrong", it is a breaking change with a minor version number that is undocumented and is not mentioned in the documents. 

Since it is to be expected that other systems use the NodeId field as lookup key, we recommend reverting the new behaviour and having it as an opt-in flag (since the old behaviour was actually odd anyway).

cristipogacean, 2022-03-25:
@derSchtefan, that's right. the background for this change can be found here:  https://github.com/Azure/Industrial-IoT/issues/1385 respectively https://github.com/Azure/Industrial-IoT/pull/1462. We considered the behavior in 2.8.1 (and less)  a bug therefore we fixed it. basically the telemetry will use for nodeId exactly the string configured in the pn.json.
You can still get the same behavior in telemetry if you change the node format in pn.json configuration to something like: `http://COMPANY.com/MS/Unit#i=6076` (namespace, hash, id).

derSchtefan, 2022-03-25:
We understand why you did this. It is just a bit odd to introduce a change that breaks E2E functionality in a 2.8.1 to 2.8.2 minor release. We also don't see it back in the release notes.

hansgschossmann, 2022-06-21:
We decided not to change this behaviour, but will document the change.

hansgschossmann, 2022-07-05:
This change is now documented at: https://github.com/Azure/Industrial-IoT/blob/main/docs/release-announcement.md#backwards-compatibility-notes

closed 2022-07-05 by hansgschossmann