Azure/Industrial-IoT #1657: Data publish frequency

Ilheu, 2022-03-27:
**I'm trying to publish data with the same frequency (every 1 second or every 100 milliseconds or every 200 milliseconds). I need to send data even it doesn't change. So I tried to use heartbeat feature it doe not work properly.**

I tried to set  "--hb=10" in Container Create Options, but it does not send the data every 10 seconds,
I tried to set  ""HeartbeatInterval": 10," in every node but also does not send every 10 seconds.

**I also want to ask what is the minimum value of heartbeat?**

cristipogacean, 2022-03-28:
There is no minimum value for heartbeat.

Please note though the --hb/heartbeatinterval are expressed in seconds. if you want to use milliseconds, you need to use  --DefaultHeartbeatInterval instead (e.g. `--DefaultHeartbeatInterval=00:00:00.200`) 

We recommend that the HeartbeatInterval is a multiple of PublishingInterval. Please take in the consideration also proper configuration of the SamplingInterval respectively QueueSize for your application.

Could you please provide more details on your issue? I just tested heartbeat on 2.8.2 and everything works as expected.

cristipogacean, 2022-03-28:
Yes, when HeartBeat is enabled, the most recent data-change will be resent with the HeartbeatInterval cycle if the value is static.

If you don't specify publishing or sampling intervals, the default value will be used (1000ms for both).

hansgschossmann, 2022-03-29:
@Ilheu looks like @cristipogacean answered your questions. We close this one for now. Please reactivate or open an new issue.

closed 2022-03-29 by hansgschossmann

marcschier, 2023-07-03:
In 2.9.0-preview4 we added UseCyclicRead option to read at interval even if data is not changing. If you use OPC Publisher 2.9 please let us know your feedback on this requested feature.