Azure/Industrial-IoT #1658: Breaking Change in OPC Publisher 2.8.2 "An item with the same key has already been added"

felixmondelo, 2022-03-29:
**Describe the bug**
If we have a duplicated node on publishednodes.json file, the OPCPublisher stops processing the rest of the nodes. In the previous version it doesn't happen, the OPCPublisher continues working.

**To Reproduce**
Steps to reproduce the behavior:
1. Add in publishednodes.json two nodes with the same ID and a third different node.
2. Launch the OPCPublisher, it doesn't subscribe to any node.

**Expected behavior**
Instead of stopping processing all the OPC nodes, ignore duplicate nodes and continue working.

**Desktop (please complete the following information):**
 - OS: Ubuntu/Windows

**Additional context**
The exact stacktrace:
Autofac.Core.DependencyResolutionException: An exception was thrown while activating Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine -> Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.WriterGroupMessageTrigger.
 ---> Autofac.Core.DependencyResolutionException: An exception was thrown while invoking the constructor 'Void .ctor(Microsoft.Azure.IIoT.OpcUa.Publisher.IWriterGroupConfig, Microsoft.Azure.IIoT.OpcUa.Protocol.ISubscriptionManager, Serilog.ILogger, Microsoft.Azure.IIoT.Serializers.IJsonSerializer)' on type 'WriterGroupMessageTrigger'.
 ---> System.ArgumentException: An item with the same key has already been added. Key: **<DUPLICATED NODE ID>**

hansgschossmann, 2022-03-29:
@cristipogacean, @karok2m can you pls take a look at this issue?

karok2m, 2022-03-29:
@felixmondelo, can you please add your sample `publishednodes.json` file here ?

cristipogacean, 2022-03-29:
@felixmondelo, indeed, it seems to be a regression. I can replicate this.

karok2m, 2022-03-30:
@felixmondelo, the intent here is to get the value of the same node twice with different names, right ?

b-enigma-con, 2022-08-10:
Is it correct that this published nodes schema does not work anymore in v2.8.3? I'm not able to supply a json array as mentioned above using the direct methods "PublishNodes" and "PublishNodes_V1". I receive a HTTP 500. Sending a single endpoint works as intended. I did a test with copy pasting the exact JSON as mentioned above using the direct method but to no avail... (http 500).

The reason I tried is because the original problem is still present in v2.8.3 of the OPC Publisher. My solution for now is to call the direct method multiple times, one time for each "dataset". This seems to be working, but is not very elegant. Is there a better solution in v2.8.3?

cristipogacean, 2022-08-10:
@b-enigma-con, PubishNodes_V1 only allows configuring a single endpoint per call. If you need to add more endpoints at the same time, check if [AddOrUpdateEndpoints_V1](https://github.com/Azure/Industrial-IoT/blob/main/docs/modules/publisher-directmethods.md#addorupdateendpoints_v1) is a better fit.

marcschier, 2022-12-13:
I added a test for the first input json and it works in 2.8.4.