Azure/Industrial-IoT #1662: Reason=BadCertificateUseNotAllowed 'Usage of certificate is not allowed.'

srjhunjhunwalacorp, 2022-04-08:
**Describe the bug**
While connecting with a OPC UA server I am getting a error "Reason=BadCertificateUseNotAllowed 'Usage of certificate is not allowed.' ", I have used "--aa" parameter already to trust the certificate, but still connection not working, tried moving the cert from rejected/certs to trusted/certs also but results are same.

**Expected behavior**
Connection successful as auto trust.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OPC Publisher - 2.8 LTS
 -  EFLOW

cristipogacean, 2022-04-08:
@srjhunjhunwalacorp, this typically means the peer's certificate, in this case the server's certificate, is not a valid/compliant OPC UA Certificate. You can find more details on the format of compliant application instance certificates [here](https://reference.opcfoundation.org/v104/Core/docs/Part6/6.2.2/). You may want to try a different generic OPC UA Client like Unified Automation's UA Expert or Softing's dataFEED OPC UA client to check the server's behavior. If you experience a similar issue, please contact the server vendor for further support.

Please note that the publisher's --aa (AutoAcceptUntrustedCertificates) argument only works for peer certificates that are OPC UA compliant and they pass the formal validation checks. Apparently in this case it is not true, therefore the publisher cannot connect.

hansgschossmann, 2022-06-21:
@srjhunjhunwalacorp since there was no response i assume the issue got resolved. feel free to reactivate if more help required.

closed 2022-06-21 by hansgschossmann