Azure/Industrial-IoT #1668: Deployment of OPC Publisher module fails.

OttoKarlsson, 2022-04-21:
**Describe the bug**
I wanted to try and do some work with OPC Publisher but ran into a problem when doing the configuration of the module on the Azure Portal. I followed this documentation on "Specify Container Create Options in the Azure Portal", link: https://github.com/Azure/Industrial-IoT/blob/main/docs/modules/publisher.md.
However, when I click "review and create" the module fails to deploy to the device.

**To Reproduce**
Steps to reproduce the behavior:
1. Create IoT Edge Device on Azure Portal
2. Install IoTEdge on a Ubuntu 20.04 Virtual machine
3. Set modules on device -> Azure Marketplace module -> OPC Publisher
4. Specify Container Create Options according to the documentation : https://github.com/Azure/Industrial-IoT/blob/main/docs/modules/publisher.md
5. Click "Create and Review"
6. See Error

**Logs**
Logs from "Troubleshoot"-tab in Azure-Portal:

<3> 2022-04-21 06:21:55.426 +00:00 [ERR] - Edge agent plan execution failed.
System.AggregateException: One or more errors occurred. (Error calling Create module OPCPublisher: Could not create module OPCPublisher
	caused by: Could not create module OPCPublisher
	caused by: invalid mount config for type "bind": bind source path does not exist: /opcpublisher)
 ---> Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException- Message:Error calling Create module OPCPublisher: Could not create module OPCPublisher
	caused by: Could not create module OPCPublisher
	caused by: invalid mount config for type "bind": bind source path does not exist: /opcpublisher, StatusCode:500, at:   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2020_07_07.ModuleManagementHttpClient.HandleException(Exception exception, String operation) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2020_07_07/ModuleManagementHttpClient.cs:line 235
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Versioning.ModuleManagementHttpClientVersioned.Execute[T](Func`1 func, String operation) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/versioning/ModuleManagementHttpClientVersioned.cs:line 155
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2020_07_07.ModuleManagementHttpClient.CreateModuleAsync(ModuleSpec moduleSpec) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2020_07_07/ModuleManagementHttpClient.cs:line 102
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 67
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 67
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 87
   --- End of inner exception stack trace ---
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.<>c.<ExecuteAsync>b__7_0(List`1 f) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 115
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 116
   at Microsoft.Azure.Devices.Edge.Agent.Core.Agent.ReconcileAsync(CancellationToken token) in /mnt/vss/_work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/Agent.cs:line 208

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

cristipogacean, 2022-04-22:
@OttoKarlsson, I believe the issue here is that the folder /opcpublisher does not exist on the host machine. Probably you need to create it manually in order to be properly mapped into the container. or you can map an already existing folder on the host system

OttoKarlsson, 2022-04-25:
> @OttoKarlsson, I believe the issue here is that the folder /opcpublisher does not exist on the host machine. Probably you need to create it manually in order to be properly mapped into the container. or you can map an already existing folder on the host system

@cristipogacean, That works, however the documentation clearly states that the IoT Edge Runtime will create the folder if it doesn't exist.

PR #1673 (2022-04-27): Doc: Fixed documentation to require mount source directory to exist on host filesystem.
Resolves #1668
closed 2022-04-27 by karok2m