Azure/Industrial-IoT #1670: Opc publisher truncate decimal points during data ingestion

anil-k-gopalakrishna, 2022-04-22:
**Describe the bug**
I have deployed edge runtime together with opcpublisher, subscribing values from different ignition OpcUa server endpoint URLs. It is observed that, for the float data type tags with  more decimal points (more than 6), the values are getting truncated while sending to Hub. Due to this, on the DB, we are seeing the same values, as the changes on the lower decimal values are not visible.
I have tested the same with  opcpublisher 2.5.4 and 2.8.1 versions.
[image]

However, whenever I am using my python opc client, I am getting the full decimal values. 
[image]
    
The published nodes configuration is as below:
[image]

 - OS: Ubuntu 18.04
 - Opcpublisher Versions: 2.5.4 and 2.8.1
 -
I am using default telemetry JSON file. Please suggest if there is any telemetry configuration settings which I am missing.

hansgschossmann, 2022-04-22:
@anil-k-gopalakrishna can you pls also try with 2.8.2 (which is the latest version)?

I see that 2.8.2 emits values with more digits after the decimal points, but i use the PubSub schema for the values:

<7> 2022-04-22 14:10:14.801 +00:00 [VRB Microsoft.Azure.Devices.Device.Client] ["Enter"] Informational: "InternalClient#8622890" "OnModuleEventMessageReceivedAsync" "(transform, Message#57446246)"
<7> 2022-04-22 14:10:14.802 +00:00 [VRB ] Message received, payload: "{\"MessageId\":\"389\",\"MessageType\":\"ua-data\",\"PublisherId\":\"opc.tcp://opcplc:50000_503386CC_test001\",\"DataSetWriterGroup\":\"test001\",\"Messages\":[{\"DataSetWriterId\":\"1000\",\"SequenceNumber\":389,\"MetaDataVersion\":{\"MajorVersion\":1,\"MinorVersion\":0},\"Timestamp\":\"2022-04-22T14:10:14.7924874Z\",\"Payload\":{\"Float\":{\"Value\":0.0016408524,\"SourceTimestamp\":\"2022-04-22T14:10:13.8754438Z\",\"ServerTimestamp\":\"2022-04-22T14:10:13.8754463Z\"}}}]}"

marcschier, 2023-01-16:
Close as duplicate to #1709

closed 2023-01-16 by marcschier