Azure/Industrial-IoT #1677: Default sampling interval should be set to 50% of the default publishing interval

hansgschossmann, 2022-04-28:
**Describe the bug**
The documentation states:
        DefaultPublishingInterval=VALUE
                                  Default value for the OPC UA publishing interval of OPC UA subscriptions
                                  created to an OPC UA server. This value is used when no explicit setting
                                  is configured.
                                  Alternative: --op, --opcpublishinginterval
                                  Mode: Standalone only
                                  Environment variable type: time span string {[d.]hh:mm:ss[.fffffff]}
                                  Alternative argument type: integer in milliseconds
                                  Default: {00:00:01} (1000)

        DefaultSamplingInterval=VALUE
                                  Default value for the OPC UA sampling interval of nodes to publish.
                                  This value is used when no explicit setting is configured.
                                  Alternative: --oi, --opcsamplinginterval
                                  Mode: Standalone only
                                  Environment variable type: time span string {[d.]hh:mm:ss[.fffffff]}
                                  Alternative argument type: integer in milliseconds
                                  Default: {00:00:01} (1000)

The default sampling interval should not be equal to the publishing interval but only half of it to prevent aliasing effects when using the default values.

karok2m, 2022-05-11:
I think that we need to think about sampling interval coupled with queue size. Default queue size is `1` at the moment. So having sampling interval be half of publishing interval might cause dropped values.

hansgschossmann, 2022-06-21:
@karok2m but keeping it the same has taking the nyqist theorem into account more probability to miss value changes.
cc: @cristipogacean

marcschier, 2022-12-12:
We should rather set default queue size to 0 = let server decide. Publishinterval is latency related, maybe we could find a heuristic taking in queue size and sampling rates in the subscription, but it is also related to the number of subscriptions in the session.  

If you want to avoid aliasing you need to oversample.  Here we could read the minimum sample rate from the variable and use it as default instead of 1 sec.  But changing any defaults breaks existing deployments.

closed 2023-07-03 by marcschier